adonno/tagreader #3: - Add condition to read only once every 5 seconds

magnusoverli, 2020-09-14:
https://github.com/adonno/tagreader/blob/1f0f8929703d0f096e15be2caaf9c442f00130dc/tagreader.yaml#L178-L178

adonno, 2020-09-14:
this can be extended. What if I want an action to happen as long as the Tag is ON the reader? 

like play a Spotify album until I remove the tag

nardev, 2020-09-15:
> this can be extended. What if I want an action to happen as long as the Tag is ON the reader?
> 
> like play a Spotify album until I remove the tag

Such feature is not so practical for the device. Triggering antenna constantly is not something hardware is made with.

Such feature is available with for example iButtons. There is iButton option with magnet that holds the tag/token in place with the intention that certain function is only available while tag is present. But it's completely different technology, there is not antenna but a direct chip contact etc.

Also, 5 seconds it too long. 2s is ok and safe.

adonno, 2020-09-15:
> Such feature is not so practical for the device. Triggering antenna constantly is not something hardware is made with.
>
>
>
> Such feature is available with for example iButtons. There is iButton option [...]

In the current config, we ask every 2 secs for an update 

That is an issue since it triggers every time when the card is on the device. 

I want it to keep the state as long as it reads the same tag id... every 2s

Landrash, 2020-10-02:
Save the last value read for a certain time and if in "continous mode" discard any duplicate values until a timeout has passed?

mccoy88f, 2020-11-05:
> Save the last value read for a certain time and if in "continous mode" discard any duplicate values until a timeout has passed?

I'm thinking about the same idea. Please add it!

jesserockz, 2020-11-05:
I have already made this change in ESPHome itself, a tag will only be reported once per scan, you have to remove it and put it back after 1 cycle for it to be reported in `on_tag` again.

closed 2021-12-25 by adonno