adonno/tagreader #5: Cannot buy a reader

kyotodesertfox, 2020-09-15:
It appears that your Cart may not be working correctly. When I select the item + colors of the reader, and then add to cart, it takes me to the Cart, which says it is empty.

Are there other ways of purchasing a reader?

magnusoverli, 2020-09-15:
Hey! There are currently some technical issues with the cart. We are working on it and hope to have a solution soon. For now, please leave Andrea a message on Twitter, and he will be in touch.

kyotodesertfox, 2020-09-15:
Great! I will keep an eye on things.

Glad you guys are already aware!

adonno, 2020-09-15:
Here is the new backup store https://adonno-crafts.myshopify.com/

kyotodesertfox, 2020-09-15:
@adonno - in the description:

> (does not work with the mobile phone apps)

What specifically does this mean?

adonno, 2020-09-15:
HA mobile apps can write tags to be used with the phones the reader uses the tag ID (for now), some tags are versatile and can do both, the tags I got unfortunately are read-only it seems

kyotodesertfox, 2020-09-15:
I went ahead and submitted an order; look forward to playing with some new toys!

closed 2020-09-15 by kyotodesertfox