adonno/tagreader #23: Unable to order on website !

vincegre, 2020-09-20:
Sorry for noise as it's not directly relevant to code published here but as there is absolutely no contact info (email or form) on your website, how do I order the boxes you offer for the tagreader as your shop is not working in Firefox ! I keep adding products at cart, I get Product added message showing up but when I access cart it's always empty !

Thanks

Vincèn

adonno, 2020-09-20:
Hey Vincen, 

Yeah the store on the website is broken I will take it offline today.

You can order your reader here https://adonno-crafts.myshopify.com

BR

vincegre, 2020-09-20:
> You can order your reader here https://adonno-crafts.myshopify.com

Thanks for link, unhappy you accept only Paypal which I completely refuse to use so I guess I'll have to download STL files and print it myself ;)

Thanks for your quick answer, it's appreciated ;)

Vincèn

adonno, 2020-09-20:
unfortunately, I have no other way. what payment method would you like to use?

adonno, 2020-09-20:
> > You can order your reader here https://adonno-crafts.myshopify.com
> 
> Thanks for link, unhappy you accept only Paypal which I completely refuse to use so I guess I'll have to download STL files and print it myself ;)
> 
> Thanks for your quick answer, it's appreciated ;)
> 
> Vincèn

I've added Apple pay and credit card

closed 2020-09-20 by adonno

vincegre, 2020-09-21:
> I've added Apple pay and credit card

Thanks for CC ;) Order passed !