adonno/tagreader #29: Use Hardware spi

Marlor, 2020-09-23:
Esphome supports Hardware SPI why not use it?

the only thing you have to do is use the dedicated pins:
```
spi:
  clk_pin: D5
  mosi_pin: D7
  miso_pin: D6
```
then Esphome will use hardware SPI:
```
[12:00:12][C][spi:081]: SPI bus:
[12:00:12][C][spi:082]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[12:00:12][C][spi:083]:   MISO Pin: GPIO12 (Mode: INPUT)
[12:00:12][C][spi:084]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[12:00:12][C][spi:085]:   Using HW SPI: YES
```
cs_pin can stay D3
buzzer can stay D8
but LED needs a new pin

magnusoverli, 2020-09-23:
What advantages are there using hardware SPI over the method currently used?

Marlor, 2020-09-23:
most users will probably not see a difference

less work for the esp
more stable in edge cases
etc

a few small reasons to use it
but no reason not to use it

adonno, 2020-09-23:
Will give it a try, might Help with NDEF support. will Bring a breaking change though

mdonath, 2020-09-24:
For those who want to try it for themselves:

My Wemos D1 mini clone needed the following configuration in order to make hardware SPI work (flipped MOSI and MISO):
```
spi:
  clk_pin: D5
  mosi_pin: D6
  miso_pin: D7
```
This resulted in exactly the same output as above (MISO = GPIO12 and MOSI = GPIO13).

PR #60 (2020-11-05): Utilize pn532_i2c from ESPHome dev
This moves to use `pn532_i2c` which is merged into ESPHome `dev` and will be released with `1.16.0` (unknown timeframe on that though)

closes #29 
closes #34
closes #44 
closes #56
closed 2020-11-05 by adonno