adonno/tagreader #34: D1 Mini won't boot with PN532 power connected

sevlitfest, 2020-09-25:
Initially had same issue on a NodeMCU - then realised it was underpowered for the PN532.
Switched to a D1 Mini - but same, it won't boot to ESP Home when the power is connected.
Boots in fine when Power and GND disconnected - but then obviously can't wake the PN532

Any ideas?

Checked Cables and Pins.

adonno, 2020-09-27:
yeah, we noticed that the PN532 might be messing with the D1. So we (actually @jesserockz ) are currently working on a new version based on I2C that should solve the issue.

lubbertkramer, 2020-09-29:
Any idea for a workaround in the time being?

tdroza, 2020-09-29:
I've been battling with this issue for days! So glad it's not just me! I can see in debug from the D1 that it boots, scans for wifi, attempts to connect then resets the wifi and starts again. Very occasionally it will connect to wifi but then doesn't connect to ESPHome (I'm using the Home Assistant plugin on an RPi 4). I had to remove the `baud:` option from the logger to get log output from the D1 over USB.

I even tried 2 different pn532 boards and about 4 different D1 Minis! I've tried powering the PN532 separately (not from the D1) but get the same issue. I can see tags being scanned in the D1 log output, but obviously Home Assistant doesn't receive an event because the D1 isn't connected. I've stripped down my yaml to remove everything but the SPI and PN532 but no luck. As soon as I disconnect the PN532 the D1 will immediately connect to WiFi.

I considered trying to set it up to use I2C but wasn't sure if it would work. I might give that a go myself while waiting for the updated version.

adonno, 2020-09-29:
@tdroza I2C won't work, for now, @jesserockz is currently working on a new library and is investing a lot of work in this. I would suggest being patient since I don't want to come up with "half" solutions. But yeah we are coming with I2C. SPI is having too many issues and messing with WIFI. I will check with Jessee if we can post a binary for tests

tdroza, 2020-09-29:
Ah ok thanks for the update @adonno. I've got my D1 Mini and PN532 on a breadboard at the moment so if you want me to test any binaries give me a shout as I can change the wiring quickly.

jesserockz, 2020-09-29:
Hello, we have been trying out using I2C, (alongside with some changes to read the NDEF formatted cards that the mobile apps can write to) in #45. You can rewire for I2C and give that PR/branch a try and let us know. 
Thanks

sevlitfest, 2020-10-16:
Just tried the new #45 and sadly still have the same issue where ESPHome doesnt seem to boot with the reader connected.
Included the ndef.h and updated the yaml but no joy :(

jesserockz, 2020-10-16:
@sevlitfest can you try #52 please?
Also does it boot if you remove the pn532 lines from yaml, but have the board attached still?

OliPassey, 2020-10-17:
I'm getting errors when trying to compile the new tagreader.yaml in ESPHome.
Using ESPHome 1.15.2
I don't believe im using an AZDelivery D1 Mini - the board has wemos.cc v3.0.0 D1 Mini Printed on it.

When the  ESPHome is first built, with no config and the PN532 connected, it boots fine.

Current compiling error:
`INFO Reading configuration /config/tagreader.yaml...
Failed config

api: [source /config/tagreader.yaml:48]
  services:  [source /config/tagreader.yaml:49]
    - service: rfidreader_tag_ok
      then: 
        - rtttl.play: beep:d=16,o=5,b=100:b
    - service: rfidreader_tag_ko
      then: 
        - rtttl.play: beep:d=8,o=5,b=100:b
    - service: play_rtttl
      variables: 
        song_str: string
      then: 
        - rtttl.play: !lambda |-
            return song_str;
    - [source /config/tagreader.yaml:63]
      service: write_tag_random
      variables: 
        continuous: bool
      then:  [source /config/tagreader.yaml:67]
        - lambda: id(continous_writing_mode) = continuous;
        - [source /config/tagreader.yaml:68]
          while:  [source /config/tagreader.yaml:69]
            condition: 
              lambda: return id(continous_writing_mode);
            then:  [source /config/tagreader.yaml:72]
              - lambda: |-
                  static const char alphanum[] = "0123456789abcdef";
                  std::string uri = "https://www.home-assistant.io/tag/";
                  for (int i = 0; i < 8; i++)
                    uri += alphanum[random_uint32() % (sizeof(alphanum) - 1)];
                  uri += "-";
                  for (int j = 0; j < 3; j++) {
                    for (int i = 0; i < 4; i++)
                      uri += alphanum[random_uint32() % (sizeof(alphanum) - 1)];
                    uri += "-";
                  }
                  for (int i = 0; i < 12; i++)
                    uri += alphanum[random_uint32() % (sizeof(alphanum) - 1)];
                  auto message = new nfc::NdefMessage();
                  message->add_uri_record(uri);
                  ESP_LOGD("tagreader", "Writing payload: %s", uri.c_str());
                  id(pn532_board).write_mode(message);
              - [source /config/tagreader.yaml:89]
                
                Unable to find condition with the name 'pn532.is_writing'.
                wait_until:  [source /config/tagreader.yaml:90]
                  not: 
                    pn532.is_writing: 
    - service: write_tag_id
      variables: 
        tag_id: string
      then: 
        - globals.set: 
            id: continous_writing_mode
            value: false
        - lambda: |-
            auto message = new nfc::NdefMessage();
            std::string uri = "https://www.home-assistant.io/tag/";
            uri += tag_id;
            message->add_uri_record(uri);
            id(pn532_board).write_mode(message);
    - service: clean_tag
      then: 
        - globals.set: 
            id: continous_writing_mode
            value: false
        - lambda: id(pn532_board).clean_mode();
    - service: cancel_writing
      then: 
        - globals.set: 
            id: continous_writing_mode
            value: false
        - lambda: id(pn532_board).read_mode();
pn532_i2c: [source /config/tagreader.yaml:136]
  
  Component not found: pn532_i2c.
  id: pn532_board
  on_tag: 
    then: 
      - homeassistant.tag_scanned: !lambda |-
          if (!tag.has_ndef_message()) {
            ESP_LOGD("tagreader", "No NDEF");
            return x;
          }
          auto message = tag.get_ndef_message();
          auto records = message->get_records();
          for (auto &record : records) {
            std::string payload = record->get_payload();
            size_t pos = payload.find("https://www.home-assistant.io/tag/");
            if (pos != std::string::npos) {
              return payload.substr(pos + 34);
            }
          }
          ESP_LOGD("tagreader", "Bad NDEF, fallback to uid");
          return x;
      - if: 
          condition: 
            switch.is_on: buzzer_enabled
          then: 
            - rtttl.play: success:d=24,o=5,b=100:c,g,b
      - if: 
          condition: 
            switch.is_on: led_enabled
          then: 
            - light.turn_on: 
                id: activity_led
                brightness: 100%
                red: 0%
                green: 100%
                blue: 0%
                flash_length: 500ms`

OliPassey, 2020-10-17:
I think i've got a USB to Serial UART thing around here somewhere, but not sure if / how it would be any use.
But could i use it to get at logs whilst it's hanging? Or do you already know what the issue may be?

jesserockz, 2020-11-05:
@OliPassey / @sevlitfest Please look at #60 and flash using the esphome `dev` branch

OliPassey, 2020-11-05:
Thanks, i'll give it a test
Much appreciated

OliPassey, 2020-11-05:
@jesserockz - I get further with this one, it boots, i get the buzzer tone as it boots, but it doesnt read any tags. 
Nothing in the log other than connected successfully.

INFO Reading configuration /config/tagreader1.yaml...
INFO Starting log output from tagreader1.local using esphome API
INFO Connecting to tagreader1.local:6053 (10.0.0.109)
INFO Successfully connected to tagreader1.local

I'm using ESPHome Dev 
Devices have turned up in  HA, LED and Buzzer works but Tag says unavailable.

[image]

adonno, 2020-11-05:
@OliPassey would you mind taking a picture of your PN532 wiring as well as the jumper settings?

OliPassey, 2020-11-05:
How should the jumper settings be set?
I've tried a few variations, so possibly not in the right position currently.

https://photos.app.goo.gl/roMDHmvJ9qnbHnEe8
Whilst some of the soldering isn't great, i don't have any shorts and all connections are secure.

adonno, 2020-11-05:
Ok please invert the Jumpers 
1 on (up) 
2 off (down) 
restart And you should be fine

OliPassey, 2020-11-05:
Still no joy im afraid.

adonno, 2020-11-05:
What did you flash the ESP with ?
In case you used ESPHome flasher please click on "show logs" while having the D1 connected  and then hit the reset button 
[image]

OliPassey, 2020-11-05:
Using 'COM10' as serial port.
Showing logs:
[14:15:30];d[I][logger:166]: Log initialized
[14:15:30][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[14:15:30][I][app:029]: Running through setup()...
[14:15:30][C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[14:15:30][C][fastled:010]: Setting up FastLED light...
[14:15:30][C][light:097]: Setting up light 'TagReader1 LED'...
[14:15:30][D][light:265]: 'TagReader1 LED' Setting:
[14:15:30][D][light:278]:   Brightness: 100%
[14:15:30][D][light:287]:   Red=100%, Green=100%, Blue=100%
[14:15:30][C][pn532:027]: Setting up PN532...
[14:15:31][W][i2c:070]: Received NACK on transmit of address 0x24
[14:15:33][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[14:15:33][E][pn532:031]: Error sending version command
[14:15:33][E][component:092]: Component was marked as failed.
[14:15:33][D][binary_sensor:034]: 'TagReader1 Status': Sending initial state OFF

Just looking to see if i have another reader i can try with.

closed 2020-11-05 by adonno

OliPassey, 2020-11-05:
I dont have another unit to test on, does the above suggest dead hardware or something else?

adonno, 2020-11-06:
Not sure but it means that the D1 isn't communicating with the PN532, check your soldering?  Resolder everything, sometimes it's something really small. 
BTW LED and buzzer are not necessary for it to work.

PR #60 (2020-11-05): Utilize pn532_i2c from ESPHome dev
This moves to use `pn532_i2c` which is merged into ESPHome `dev` and will be released with `1.16.0` (unknown timeframe on that though)

closes #29 
closes #34
closes #44 
closes #56