adonno/tagreader #35: Version with integrated battery

atze09, 2020-09-26:
Hello Guys,
Does anyone already designed a case dir a portable Version with integrated Power source.
I would like to build a juke Box for my daughter. 
Integrated would Like more nice than a Power Bank.

Best regards,
Dominik

adonno, 2020-09-27:
Hello, 

So the issue here would be... 

- you would need a timeout so that the D1 goes to sleep after a while not to drain the battery.
- you would probably need a battery status indicator if it's loaded or not 
- based on the above you would have to add a "power on/power off" button. 
- my knowledge isn't there yet to know the exact mah battery required to run a specific time. 

These are things that can be done though but will require some time to think about and to places in the case.  Right now we are focusing on getting the hardware perfect for the 1st version of the Tag Reader. We'll see what we can come up with in future

balloob, 2020-09-27:
Use a proximity sensor to enable NFC / connect to wifi when a card is being held close.

adonno, 2020-09-28:
you would need a physical switch. Since having a proximity sensor polling every 1 second would require the D1 to be on at all time.

tdroza, 2020-09-29:
You _could_ change the case so that instead of waving the tag over the reader, you slide it into a slot. At the back of the slot you could have a hinged contact switch so that inserting the card triggers the switch which powers up the D1. ...also the current case is a perfect size to fit a QI wireless charging receiver ;-)

jesserockz, 2020-09-29:
You could make the lid of the case become a switch so when you tag the card to it, pressing the lid/switch wakes the d1 from sleep then proceeds to scan then back to sleep.

gadgetshed, 2020-10-21:
If you are sticking with ESPHome then the chip will need to change to an ESP32.  ESPHome only supports deep sleep currently and not a light sleep.  In deep sleep the only way to wake a D1 Mini 8266 chip is on a preset timer.  ESP32 however supports deep sleep and trigger pins so the idea of a switch to reactivate would work.

closed 2021-01-04 by atze09