adonno/tagreader #40: replace UID with NDEF

adonno, 2020-09-28:
Whenever a tag with a NDEF is scanned,  check if there is an automation for that tag, and if so, replace the UID by NDEF in the automation 

This is to avoid that a Tag would trigger 2 automations (one based on the NDEF another on the UID) at once. If users want to have a different action based on the device they use to read the tag, they will have to set it up by device.

Not sure if this should be managed on Home-Assistant side though

balloob, 2020-09-28:
We shouldn't do this. If NDEF available, we fire NDEF. Adding something like this would only make it more complicated for users in the end.

adonno, 2020-09-28:
This would then be a breaking change for users who have already set up an automation based on UID then ?

balloob, 2020-09-28:
But they flashed an NDEF value on it themselves?

adonno, 2020-09-28:
ok makes 100% sense, will have to be mentioned in the release notes.

closed 2020-09-28 by balloob