adonno/tagreader #44: Unstable WiFi connection

tipofthesowrd, 2020-09-29:
The WiFi connection of the esp8266 to the network is unstable.
When connecting the pn532 module and after the setup of the module the WiFi connection and connection to the API stops & hangs often as evidenced by missing pings.

When the module is (physically) connected in HSU mode (or not setup) the connection is stable.

gadgetshed, 2020-09-29:
I can concur that the network connection is very flaky and it seems to drop off the network frequently.

gadgetshed, 2020-09-29:
I'm running a spi and i2c version next to each other and the i2c version is very stable in comparison so it's definitely the way to go.

As an aside the new schematic has the buzzer on D7 but the yaml is on D3 so I'm using D3 for wiring.

tipofthesowrd, 2020-09-29:
Where can I find the i2c version? Esphome didn't seem to have it in the
docs?

jesserockz, 2020-09-29:
@tipofthesowrd #45

tipofthesowrd, 2020-09-30:
I tried the i2c version, altough I see some improvement, there seem to be some other issues (see below).

As for the WiFi instability, I have the impression it is caused by the proximity of the pn532!!!
I'm using the printed STL files, when Wemos & the PN532 are close to each other the WiFi issues are there.
When I add jumper wires to seperate the boards without the printed housing the connection is stable.

As for the NDEF / I2C version: 

Tags written by the Home Assistant Android app are detected either as a NDEF and other times with a UUID.
Example from the same tag read in succession. This causes Home Assistant to detect it as two different tags!
Since the I2C version is not yet official I didn't want to open a seperate issue.

```
[18:28:00][D][ndef_tag_reader:057]: Tag Found!
[18:28:00][D][ndef_tag_reader:102]: No NDEF
[18:28:00][D][ndef_tag_reader:105]: Tag UID: 29-79-7B-9D
[18:28:00][D][text_sensor:015]: 'TagReader Tag': Sending state '29-79-7B-9D'
[18:28:01][D][text_sensor:015]: 'TagReader Tag': Sending state ''
[18:28:42][D][ndef_tag_reader:057]: Tag Found!
[18:28:42][D][ndef_tag_reader:065]: Tag has NDEF
[18:28:42][D][ndef_tag_reader:068]: Tag has 1 records
[18:28:42][D][text_sensor:015]: 'TagReader Tag': Sending state '8d2a73fc-fa4f-4968-8e3e-21e3b4034bbe'
```

adonno, 2020-09-30:
hey @tipofthesowrd what is the brand of this D1 you are using?  We seem to have issues with AZdelivery branded boards

tipofthesowrd, 2020-10-01:
@adonno  Nothing special, they are unbranded, ordered from a local seller for this projet. 

I was checking some more and found that when the boards are separated the ping only looses one packet every 10 seconds.
When placed in the housing, almost no ping comes through, except when the PN532 is not setup at boot. 

I was looking over the datasheet of the pn532 if there are some RF settings to tweak.

jesserockz, 2020-10-01:
I too have looked over the user guide for the board and could only find the RF on/off command, but nothing to specify power level.

tipofthesowrd, 2020-10-01:
No, I couldn't find anything either. However, there might be something in the way the library operates.

I noticed when reading a tag the ping gets better. I just found this:

```
  if (report) {
    char buf[32];
    format_uid(buf, nfcid, nfcid_length);
    ESP_LOGD(TAG, "Found new tag '%s'", buf);
  }

  this->turn_off_rf_();
```

After reading a tag the RF field is turned off, which might explain why I'm seeing a stable connection for a time after reading a card.

mories76, 2020-10-01:
Hi all,

It looks like I am experiencing the same wifi stability issues with a genuine lolin wemos d1 mini.
The tags read fine, HA is constantly connecting and disconnecting.
The PN532 is connected with a few centimeters of cable.

Later today I will solder some header and use jumper cables and see if the distance between the two have an effect.

mories76, 2020-10-02:
I have to withdraw my last comment ;-)
My wifi was having issues, I was too quick with my with my conclusion.

tipofthesowrd, 2020-10-05:
It might be with my specific type of D1 mini board: it is labeled DOITING. 
Reference: https://geekenargentina.wordpress.com/2019/10/08/esp8266-connectivity-troubleshooting/

When touching the antenna, the connection is immediately available. I'll see if I can desolder the tin can and check the capacitor.

tobiasbayer, 2020-10-11:
I can confirm it has something to do with the case or the position of the components inside it. My WiFi connectivity issues disappear as soon as I get the components out of the case and pull the ESP and the PN some (about 5) centimeters apart.

tipofthesowrd, 2020-10-11:
What's the marking on your D1 Mini case? I opened one case and removed the
capacitor as in the reference I linked above.

It's slightly better but still I'm getting serious and regular long
disconnects. I'm waiting for some more to arrive from AliExpress hoping
they're from another manufacturer

tobiasbayer, 2020-10-11:
[image]

adonno, 2020-10-11:
@tobiasbayer I'd like you to test an Esphome Native I2C based binary  :  [tagreader-i2c.zip](https://github.com/adonno/tagreader/files/5362204/tagreader-i2c.zip)
same goes for everyone else having issues. 

**You will have to change the wireing for I2C**
```
D1 --> SCL
D2 --> SDA
```
(please also use GND and VCC from the I2C side)
And do change the Jumper switches on the PN532 for I2C ( 1 Up / 2 Down)

I've been testing for several days and I must admit it works almost flawless (disconnected sometimes due to me moving it around)
[image]

tipofthesowrd, 2020-10-12:
@adonno I just flashed this verison. I'll report back about my findings.  
What's specific about this binary? Anything different the pull request by @jesserockz ?

jesserockz, 2020-10-12:
It's a binary from along the track of #52.
#52 has tag writing as well, but that binary only has reading

tobiasbayer, 2020-10-12:
I flashed the I2C version and did the rewiring. WiFi seems to be pretty stable now.

tipofthesowrd, 2020-10-12:
Same here! This specific binary works great for me as well.

dojf, 2020-10-13:
I flashed the I2C version on a extra ESP8266 D1 Mini and extra PN532 . The difference is big as can be seen in the comparison  below (same timespan). Great work, thank you!

[image]

Also a very pleasant surprise that the (programmeble) long tags work in this version, thanks!

slemiere, 2020-10-16:
Same effect for me, I was getting disconnects every few minutes and now with the I2C build it's super stable. D1 mini is coming from az-delivery

dartfrogdk, 2020-11-04:
this file seem very stable

closed 2020-11-05 by adonno

apeeters, 2020-11-24:
> I noticed when reading a tag the ping gets better. I just found this:
> ```
>   this->turn_off_rf_();
> ```
> 
> After reading a tag the RF field is turned off, which might explain why I'm seeing a stable connection for a time after reading a card.

I was the one who added that line in an attempt to get a more stable WiFi connection. I now moved to the pn532_i2c version which is WAY more stable, great work!

PR #60 (2020-11-05): Utilize pn532_i2c from ESPHome dev
This moves to use `pn532_i2c` which is merged into ESPHome `dev` and will be released with `1.16.0` (unknown timeframe on that though)

closes #29 
closes #34
closes #44 
closes #56