adonno/tagreader #51: NFC Tags not read

tobiasbayer, 2020-10-10:
My setup does not read any NFC Tags. They simply do not show up in the "Tags" view in HA when I place them near the reader.
Bascially, the communication between the ESP and HA seems to work ok as I can turn on and off the LED via the Lovelace UI.
The PN532 seems to be powered on at least as the red LED is constantly on.

Any hints on how I could debug this?

mories76, 2020-10-10:
Do you have access to the console of the ESP ?
The console reports whether the PN532 can be found or not.

Also how did you connect your PN532 ?
spi, i2c. The dipswitches on the PN532 should be set accordingly

tobiasbayer, 2020-10-10:
I wired it up like shown [here](https://github.com/adonno/tagreader#connecting-the-components) and set the dip switches accordingly.
I'll try to get access to the console and will report back here.

tobiasbayer, 2020-10-10:
I got this output:

`...
[0;35m[C][spi:081]: SPI bus:[0m
[0;35m[C][spi:082]:   CLK Pin: GPIO16 (Mode: OUTPUT)[0m
[0;35m[C][spi:083]:   MISO Pin: GPIO5 (Mode: INPUT)[0m
[0;35m[C][spi:084]:   MOSI Pin: GPIO4 (Mode: OUTPUT)[0m
[0;35m[C][spi:085]:   Using HW SPI: NO[0m
...
[0;35m[C][pn532:355]: PN532:[0m
[0;35m[C][pn532:367]:   CS Pin: GPIO0 (Mode: OUTPUT)[0m
[0;35m[C][pn532:368]:   Update Interval: 2.0s[0m
[0;35m[C][rtttl:017]: Rtttl[0m
...
[0;36m[D][api.connection:604]: Client 'Home Assistant 0.116.2 (192.168.178.111)' connected successfully![0m
[0;36m[D][rtttl:093]: Playback finished[0m
[0;32m[I][ota:046]: Boot seems successful, resetting boot loop counter.[0m
`

mories76, 2020-10-10:
What happens when you offer a tag, do you see anything happening on the console ?
It should include something like below, depend on the version of pn532 integration your running.
[16:00:17][D][pn532:150]: Found tag '04-7A-CC-1A-F5-48-80'

Do you have different tags available ? Can you scan the tag with your phone, with a tool something like NFC Tools.
I have one type of tags which isn't read yet with the firmware I am running. Mifare Ultralight

tobiasbayer, 2020-10-10:
Ok, it now worked and shortly stopped working after I moved the reader and worked again after moving it some more. I suspect some sort of bad soldering or something like that. I'll redo the soldering.
Thanks a lot for your help debugging this!

closed 2020-10-10 by tobiasbayer

RosaEinhorn, 2021-02-23:
Hello, I have the same problem. Tags are not read and I am getting
```
[10:23:57][W][i2c:070]: Received NACK on transmit of address 0x24
[10:23:58][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[10:23:58][E][pn532:019]: Error sending version command
```
in the logs.

The soldering is not perfect, but it should work (see pictures).

[image]
[image]

Would appreciate any help. :)