adonno/tagreader #53: Need help to use your code with Arduino Uno + Ethernet Shield and Wiegand26 NFC reader

bennyboom38, 2020-10-14:
Hello

I can't use wifi to connect my NFC reader to Home-Assistant.
I would like to do it with Arduino Uno + Ethernet Shield but I don't know how to change the code to work with HA?
Could you help me how it works with ESPHOME to adapt for arduino uno and ethernet shield please?

Thanks a lot,

Regards

Ben

adonno, 2020-10-15:
There are two ways to approach your issue. 

Running ESPHome on an Arduino won't be possible AFAIK. So what you can do, 
- use don's NDEF library for Arduino: https://github.com/don/NDEF/ and send the data to HA via MQTT
- use an ethernet enabled ESP32 board https://esphome.io/components/ethernet.html and continue using ESPHome

bennyboom38, 2020-10-15:
Thank you for your answser :)
I already have a part of code that returns this when I scan a tag : 
```
Bits=26
RAW Binary=01011000101101000011010100
```
But I don't understand how to send it to Home-Assistant (by api ? http://my-home-assistant-IP:8123/api/....) to allow Home-Assistant to get the tag id and compare to the list of allowed tags to trigger automation to open or not the door

Thanks a lot for your help

Regards

jesserockz, 2020-10-18:
Hi @bennyboom38 
I don't think there is an API call to notify Home Assistant of a tag scan, as it is done via the integrations at the moment.

Also, as this is completely unrelated to this project, I would suggest that you ask on the [Home Assistant Community forums](https://community.home-assistant.io), or the [Home Assistant Discord server](https://www.home-assistant.io/join-chat/).

Jesse

bennyboom38, 2020-10-20:
Hello Jesse
Thanks for your answer, I'm going to check with HA community

Regards

closed 2020-10-22 by jesserockz