adonno/tagreader #54: No scan after a few minutes

dcjona, 2020-10-17:
Hi,

I made the prototype and test it. everything is ok but after a few minutes i hear the buzzer and then no more scan can be done.
Like if it shutdown ou hibernate...

I have to launch the show logs on esphome flasher to wake up ... else nothing works

Could you help ?

jesserockz, 2020-10-18:
Hi @dcjona 

We are going to need more information than that to be able to help...
The only way to know what is going on, is to reproduce the issue while watching the serial logs, otherwise there is no way t know what is happening.

Jesse

dcjona, 2020-10-21:
Hello, i made another module and same issue..

i don"t know why when i power off esphome flasher the tag reader doesn't works anymore...

jesserockz, 2020-10-22:
Thanks @dcjona can you try using the instructions, code and wiring configuration in #52?

The wake up failed is a bit of a weird error with the current pn532 code and I have a branch in esphome which gives it a few upgrades and more reliability.

dcjona, 2020-11-26:
thqnk you @jesserockz sorry for the delay i will try

seranoo, 2021-05-10:
I have had issues with the i2c communication failing after a few minutes. Got a read error, only thing to do was to power cycle the ESP. 

I was initially using a Wuqiong ESP-WROOM-32 (https://www.amazon.se/gp/product/B086ZQ7TL5/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1), i.e. this was the ESP that I had the above issues with. 

I then changed to a 
WEMOS/LOLIN D1 mini v3.1.0 (https://www.m.nu/esp8266/d1-mini-v310) and now everything works perfectly. It has now been running for a couple of days, no issues. I read somewhere that it potentially could be some issue with "missing" pullup resistors on the i2c GPIOs, haven't tested that yet.

BTW, I use a PN532 from here: https://www.amazon.se/gp/product/B0919BP156/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

I run the setup without the buzzer and the LED, works super nice!

closed 2021-09-17 by adonno