adonno/tagreader #57: STL printing in two colors

chadlyy, 2020-10-28:
Do you have any recommendations on how to print your STLs in 2 colors? It looks like I need 2 parts to merge.  Sorry, I'm really new to 3d printing, any help would be really appreciated!

zuidwijk, 2020-10-28:
you can try this way: https://youtu.be/1nBnVtOEAiY

zuidwijk, 2020-10-28:
I'm printing a case myself too. And just printed case (bottom part). Tomorrow I'll try to print the top cover in two colors the way shown in the video.

zuidwijk, 2020-10-29:
Ok I tried two ways:

1) as described in the youtube video. That didn't work. It paused and after 1 second resumed immediately. These are my settings:[image]
2) used the filament change script (M600 code). That didn't work on my CR-6 SE. 

So for me, I'm still searching for the right way. For now I've got two white covers. 
[image]

adonno, 2020-10-30:
Hello I used that exact same Video but I also Use octoprint 
I am using an Ender5 Pro

zuidwijk, 2020-10-31:
> Hello I used that exact same Video but I also Use octoprint 
> 
> I am using an Ender5 Pro 

For the Ender it's supported, yet for the CR-6 SE not, unfortunately.

zuidwijk, 2020-11-01:
Not tried yet, but the right code should me M25. You can modify the code and change M0 to M25, you can also add it like this:
[image]
````
;TIME_ELAPSED:816.988185
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 2
M83 ; switch to relative E values for any needed retraction
G1 F3000 E-25
G1 F300 Z1.6 ; move up a millimeter to get out of the way
G1 F9000 X20 Y20
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S25 ; standby temperature
M0 ; Do the actual pause
M25
M109 S200 ; resume temperature
G1 F3000 E25
G1 F3000 E-25
G1 F300 Z1.6
G1 F9000 X106.468 Y74.537
G1 F300 Z0.6 ; move back down to resume height
G1 F3000 E25
G1 F1500 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E499.70827
````
More explained here: https://www.youtube.com/watch?v=mnujJOvzf0o

zuidwijk, 2020-11-01:
Ok just tried it. It works... but you've got to be careful. The change of filament didn't went that smooth and when started printing, the 2nd change (back to white) made a big blob of filament which I could peel away with tweezers just before the nozzle printed again in the same area. [image]
[image]
[image]

adonno, 2020-11-01:
I avoid them by staying there and pull the blob with the tweezers

closed 2020-11-20 by adonno