adonno/tagreader #59: Automatic Enable Buzzer and/or Led Light

mccoy88f, 2020-11-05:
Thanks for all your great work!
Is possible to enable directly in esphome code the buzzer and/or the light?
Because every time it reboot i need to reenable it on ha.
Thanks

jesserockz, 2020-11-05:
You can add these into the on_boot sequence, probably at the top of it:
```yaml
  on_boot:
    priority: -10
    then:
    - switch.turn_on: led_enabled
    - switch.turn_on: buzzer_enabled
    - wait_until:
      ...
```
And then they will be enabled all the time after a reboot.

adonno, 2020-11-05:
@jesserockz should we maybe add info about this in readme?  like 

buzzer and LED are disabled at boot if you wish to have them enabled, please add 
```  yaml
  on_boot:
    priority: -10
    then:
    - switch.turn_on: led_enabled
    - switch.turn_on: buzzer_enabled
    - wait_until:
      ...
```

jesserockz, 2020-11-05:
Good idea =)

RichieFrame, 2020-11-10:
I actually added this to mine, to show it was waiting for the API connection, vs having no visual feedback about what was going on:

```
  on_boot:
    priority: -10
    then:
    - light.turn_on:
        id: activity_led
        brightness: 20%
        red: 100%
        green: 0%
        blue: 0%
    ...
```

As for auto enable buzzer and led light, I would suggest using a helper toggle in HA, then have esphome pull the variable once the api is connected, that way you do not need to reflash to change the default... I am coding this now and will test how well it works

RichieFrame, 2020-11-10:
You know... a better option may be to use the restore_state variable for those switches

```yaml
switch:
- platform: template
  name: "${friendly_name} Buzzer Enabled"
  id: buzzer_enabled
  icon: mdi:volume-high
  optimistic: true
  restore_state: true
- platform: template
  name: "${friendly_name} LED enabled"
  id: led_enabled
  icon: mdi:alarm-light-outline
  optimistic: true
  restore_state: true
```

RichieFrame, 2020-11-10:
restore_state did not seem to work at all, I assumed it would get the last active state from HA and set it, but nothing happened.

helper toggle on the other hand works:

```yaml
  on_boot:
    priority: -10
    then:
    ...
    # set defaults for led and buzzer
    - if:
        condition:
          binary_sensor.is_on: defaults_buzzer
        then:
          - switch.turn_on: buzzer_enabled
    - if:
        condition:
          binary_sensor.is_on: defaults_led
        then:
          - switch.turn_on: led_enabled
```
```yaml
binary_sensor:
  - platform: status
    name: "${friendly_name} Status"
  - platform: homeassistant
    id: defaults_buzzer
    name: "${friendly_name} Default Buzzer Setting"
    entity_id: input_boolean.tag_reader_defaults_buzzer
  - platform: homeassistant
    id: defaults_led
    name: "${friendly_name} Default LED Setting"
    entity_id: input_boolean.tag_reader_defaults_led
```

Then create those 2 toggle helpers, set your preferred defaults, and reflash

closed 2020-11-20 by adonno