adonno/tagreader #61: Don't scan nfc tag (Installed latest code using latest esphome dev)

mccoy88f, 2020-11-06:
Hi, as object tag reader doesn't scan anything. Anymore.
Old code and esphome (not dev and not beta work)
I alse listen to buzzer previous code when connect to ha sing a Jingle (after like 10 seconds)
With new code versione only a beep just connected (after like 0 second)
Anyone have this problem?
Thanks

balloob, 2020-11-06:
Make sure you also use updated schematics.

mccoy88f, 2020-11-06:
Do i need to use this new scheme?
https://github.com/adonno/tagreader/raw/master/Schematics/tag_reader_schematics_v2.png
Readme also say:
"Also make sure that you have set the switches on the PN532 to the following:
Switch 1: On (up)
Switch 2: Off (down)"
Is it correct or i need to put all off?
Thanks

mccoy88f, 2020-11-06:
Thanks info in readme is correct. All works now

closed 2020-11-10 by jesserockz