adonno/tagreader #62: Scanning an Android phone as a tag?

poblabs, 2020-11-07:
When using my android phone as a tag, and scanning it on the scanner, Home Assistant recognizes it as a different tag each time it's scanned. 

Any tips on how to "lock down" just one tag id to my phone?

[image]

adonno, 2020-11-08:
So android will change the UID on every scan however there seems to be people and apps that mange to emulate a ndef tag 

https://stackoverflow.com/questions/6138077/can-an-android-nfc-phone-act-as-an-nfc-tag

More precisely https://stackoverflow.com/a/8240340

But also please read this post that advises to just use a sticker instead of your phone

https://stackoverflow.com/a/9149881

closed 2020-11-08 by balloob

poblabs, 2020-11-08:
Looks like it's possible as an app. Would be great to have this added to the companion app as an option. @balloob @adonno 

Thanks gents