adonno/tagreader #64: tagreader on wifi but homeassistant not detecting them

maxandersen, 2020-11-08:
I'm running latest homeassistant 1.117(?), I got two tagreaders, I configure them via their own access point and I can now see them on the network (can ping them) but homeassistant does not come up with a notification.

Any way to manually add the readers via Integration as I know their ips ?

balloob, 2020-11-08:
ESPhome is discovered by `zeroconf` integration. You can also set up the ESPHome integration directly with an IP.

closed 2020-11-08 by balloob

adonno, 2020-11-08:
Sometimes HA needs a restart for whatever reason, and it should auto discover it.

But you can add it via the integrations page by clicking on the orange plus sign on the bottom right. Try to add it by its name first, except if your reader will have a dedicated ip, then you can add it via the ip.

maxandersen, 2020-11-08:
I've updated to latest hassio / homeassistant. no change.
forced restart. no change.

ping via ip works, ping via hostname fails for most pings (despite it being same ip) 

Adding manually ESPHome does not work ...

before I try add it fully manually in the config is there a sanity check I can make of the tag readers they behave as expected?