adonno/tagreader #65: turn off red LED on PN532

alesaund, 2020-11-09:
The red led on my PN532 is always on, while the WS2812 led works fine, 
how to disable the led on the nfc reader board?

RichieFrame, 2020-11-10:
Without seeing a board schematic I would not think that is possible, the LED may just tap into the output of the power regulator on the board, so as long as there is power the light is on.

You could try to solder it off for a permanent solution

I am using the board upside down so there is less light being emitted as the board blocks more than half, you only lose about 2mm of range

adonno, 2020-11-10:
That is exact , 
You can either unsolder the LED, or take an exacto knife and cut the vcc lane, but the easiest way (and pleas try this first before mechanically modifying the board) to do this would be by using a sharpie.

alesaund, 2020-11-13:
The sharpie works great! Tnx

closed 2020-11-13 by alesaund

lnxbil, 2021-01-26:
Could you please add this to a kind of FAQ of some sort?

StefanDorschu, 2023-11-25:
I am not familiar with using a sharpie to do this. What do i exactly have to do? A picture would be fantastic.