adonno/tagreader #66: HID iClass Badge Tag scans a different UID each time

Royal2000H, 2020-11-10:
Hello,

I just received the Plug-n-Play tag scanner from Adonno - Thank you!

While I will generally use it for normal tags (like the ones included), I was hoping to also be able to use my work badge, which is a HID iClass badge. Unlike HID Prox (125 kHz), the HID iClass badges are 13.56 mHz - therefore they are read by the scanner.

The problem is that I am getting a different UID each time I scan the badge. I assume this is likely because HID iClass works on a challenge/response type of security system. But is there any hidden field that can be read by the scanner that would be constant each time from a specific HID iClass badge?

jesserockz, 2020-11-10:
I highly doubt you will reliably be able to use this. It will be generating a random UID everytime because as you mentioned it most likely does not even care about the uid, but uses a challenge/response after the initial connection is made.

Without digging deeper into how those tags work we wont be able to read/identify them with the current code. 
That would be way too much work for this project and for ESPHome itself.

kyotodesertfox, 2020-11-13:
> Without digging deeper into how those tags work we wont be able to read/identify them with the current code.
> That would be way too much work for this project and for ESPHome itself.

Is it then safe to assume that attempting to use this with phones is just not a feasible thing to do / likely wont ever happen?

I was banking on the idea of using this to unlock my front door whenever "Works with Hey Google" is fully implemented. :(

adonno, 2021-05-01:
I solved it differently, I use my phone to scan a sticker tag that is on my door. Home Assistant or iOS (in my case) do the rest

closed 2021-09-17 by adonno