adonno/tagreader #70: " Timed out waiting for readiness from PN532!"

sirvictory444, 2020-11-13:
Hello,
Changing the logger level to VERBOSE I can see the error message:  "Timed out waiting for readiness from PN532!"
Card reads do not work.  If I hold the card over the NFC sensor for 5-10 seconds, eventually a read will come in.  
I've double checked the wiring, proper soldering, etc, thats not an issue.   I know the i2c version of this is still in the dev mode.
Do I have a bad board? I've tried with 2 different D1 minis, but only have the 1 PN532.
The hardware is in the 3d printed case from this repo.
My ESP board does not say AZ Delivery.
Thank You,

RichieFrame, 2020-11-14:
Could be a bad reader board, I had 1 out of 3 not work out of the box, I was able to trace it to bad solder flow which joined 2 of the independent circuits on the board, because a component was off center by about 0.25mm when it was surface mount soldered

adonno, 2020-11-14:
In this case it's the firmware 
I would suggest using the newest yaml in combination with ESPHome dev as explained on the Readme.md you'll have to resolder to I2C instead of SPI but it will run rock solid

sirvictory444, 2020-11-14:
> In this case it's the firmware
> I would suggest using the newest yaml in combination with ESPHome dev as explained on the Readme.md you'll have to resolder to I2C instead of SPI but it will run rock solid

Yes, this is with I2C.  Using the latest firmware yaml, wiring, and dev version of ESPHome.   SDA & SCL wiring.  I suppose I'll order another board if you're saying it's not a software issue?

adonno, 2020-11-14:
last question are the Jumpers set correctly?

sirvictory444, 2020-11-14:
> last question are the Jumpers set correctly?

Yes, I0 on/up, I1 off/down.     Like I said, it does work once every 30 seconds maybe, in between getting random mismatched data.  I don't think it would do that unless it was set correct, no?  Maybe DIP switch is bad, I've seen that with even the larger through hole kind.  I'll try to use some thin wire to solder them, if I'm able to without messing it up.

sirvictory444, 2020-11-15:
I ordered another board.  I suppose we can close this ticket, as we're saying it's probably a hardware issue.

closed 2020-11-15 by sirvictory444