adonno/tagreader #71: Trigger the buzzer (with custom sounds) and / or set the light (with color) from HA

Mincka, 2020-11-14:
I just received my tag reader today. Thank you for this great project! 👍 

I would like to know if you think it could be possible to use the Notify platform / a Service in HA to send specific commands to the device.
I think it could be great to be able make it beep with custom music notes that you could send as a message, or to use its LED as a light in HA.

Since the idea is to have this device bound to a room, I would use this to make distinct sounds to warn me about specific events or status. 

A few ideas:
- Warning beep + red flash if I open the door to leave but some windows are open
- Reply based on status (1): I flash a badge to request a binary status, the reader answers with yes / no sound + green / red light.
- Reply based on status (2): I flash a badge to request the weather and the tag reader answers (from automation) with a color code (eg. red for rain, orange for chance of rain, white for snow, yellow for sunny...).
- Permanent light that changes color based on a associated sensor (e.g. ambient air status from green to red)

RichieFrame, 2020-11-14:
All these features already exist in the code. The LED is already reported as a light in HA, you can do all the normal stuff with it, although the brightness cannot go below somewhere around 18% due to limitations with the chipset pwm and the LED frequency, using a different LED on ESP32 with LEDC output allows it to go lower.

As for using the buzzer, the service is available in HA as ```esphome.tagreader_play_rtttl```

For alert notifications, I have already integrated the device into my alert flow, which is colored based on the type/severity of alert. Blue (general notification 1), Green (general notification 2), Yellow (alert, maybe you should look into it), Orange (alert, you should look into it now), Red (alert, danger)

adonno, 2020-11-14:
I didn't think about it but I really like this especially since I'm thinking about a more "enhanced" version once I get everything delivered

Mincka, 2020-11-14:
Thank you @RichieFrame! Next time, I will dig deeper before submitting "issues". I am very new to ESPHome. The platform looks like very powerful.

Indeed, it looks like I can already answer to all my ideas with your input. I will check tomorrow and close this issue. 

Maybe it could be useful to highlight a bit more these "indirect" features of the tag reader.

Mincka, 2020-11-15:
I tried to change the colors of the LED, it worked but at some point, the device became unavailable.
I needed to reboot it a few times to get it back available in HA, very strange. I tried to reproduce but didn't manage to.

For the buzzer, it works perfectly with the play_rttl service and a song_str string
`song_str: "McGyver:d=4,o=4,b=160:8c5,8c5,8c5,8c5,2b,8f#,a,2g,8c5,c5,b,8a,8b,8a,g,e5,2a,b.,8p,8c5,8b,8a,c5,8b,8a,d5,8c5,8b,d5,8c5,8b,e5,8d5,8e5,f#5,b,1g5,8p,8g5,8e5,8c5,8f#5,8d5,8b,8e5,8c5,8a,8d5,8b,8g,c5,b,8c5,8b,8a,8g,a#,a,8g."`

I'm just putting this here as reference:
https://esphome.io/components/rtttl.html

😍

closed 2020-11-15 by Mincka