adonno/tagreader #74: Piezo buzzer volume is too low | INFO

CarpeDiemRo, 2020-11-19:
There is any possibility to increase the volume of the piezo buzzer?

RichieFrame, 2020-11-20:
I am using an actively powered passive buzzer, it takes constant 3.3v input plus the 3,3v trigger from the ESP GPIO... it is quite loud. You can even add a potentiometer to the ground for a volume knob, I am using a 470ohm resistor to get appropriate volume for an open air design

CarpeDiemRo, 2020-11-20:
> I am using an actively powered passive buzzer, it takes constant 3.3v input plus the 3,3v trigger from the ESP GPIO... it is quite loud. You can even add a potentiometer to the ground for a volume knob, I am using a 470ohm resistor to get appropriate volume for an open air design

I have an active buzzer, too and it works perfectly. The major problem is that is too big and it is not fit into adonno box.

adonno, 2020-11-20:
there is someone who drilled a hole in the case to fit the Buzzer ,  unfortunately considering that we wanted to have a slim design we had to revert to this buzzer. Function follows form :-D

closed 2020-11-22 by CarpeDiemRo