adonno/tagreader #75: No NFC Tag is reading

ollimagic, 2020-11-22:
I Build everything like in the description but no tag arrived in Home Assistant NFC
i add tham vie Mobile but nothing happen when i use tham on the reader

dmeus, 2020-11-22:
Likewise, would love to learn how to fix this.

adonno, 2020-11-22:
Would you mind sending me a picture of the "Reader? " are the Jumpers correct? 
(1 up  2 Down)

dmeus, 2020-11-22:
> Would you mind sending me a picture of the "Reader? " are the Jumpers correct?
> (1 up 2 Down)

Atm they are 1 down, 2 up as stated in the readme? Connection to my network works (d1 has an ip adress) and home assistant discovered 4 entities. At least so far so good. No dice on recognising a tag tho... in the esphome log it also only states ‘api connected’, no further log. Is that normal?

adonno, 2020-11-22:
They have to be 1UP 2DOWN

[image]

That should solve the issue

ollimagic, 2020-11-22:
i guess i did everything like on your picture

ollimagic, 2020-11-22:
yes the reader appears in hassio

ollimagic, 2020-11-22:
INFO Reading configuration /config/esphome/Tag2.yaml...
INFO Starting log output from tagreader.local using esphome API
INFO Connecting to tagreader.local:6053 (192.168.178...)
INFO Successfully connected to tagreader.local
[17:42:44][I][app:105]: ESPHome version 1.16.0-dev compiled on Nov 22
2020, 16:28:28
[17:42:44][C][wifi:443]: WiFi:......
[17:42:44][C][wifi:303]:   SSID: ....
[17:42:44][C][wifi:304]:   IP Address: 192.168.178...
[17:42:44][C][wifi:306]:   BSSID: [redacted]
[17:42:44][C][wifi:307]:   Hostname: 'tagreader'
[17:42:44][C][wifi:311]:   Signal strength: -48 dB ▂▄▆█
[17:42:44][C][wifi:315]:   Channel: 1
[17:42:44][C][wifi:316]:   Subnet: 255.255.255.0
[17:42:44][C][wifi:317]:   Gateway: 192.168.178.1
[17:42:44][C][wifi:318]:   DNS1: 192.168.178.1
[17:42:44][C][wifi:319]:   DNS2: (IP unset)
[17:42:44][C][i2c:028]: I2C Bus:
[17:42:44][C][i2c:029]:   SDA Pin: GPIO4
[17:42:44][C][i2c:030]:   SCL Pin: GPIO5
[17:42:44][C][i2c:031]:   Frequency: 400000 Hz
[17:42:44][C][template.switch:058]: Template Switch 'TagReader Buzzer Enabled'
[17:42:44][C][template.switch:058]:   Icon: 'mdi:volume-high'
[17:42:44][C][template.switch:059]:   Restore State: NO
[17:42:44][C][template.switch:060]:   Optimistic: YES
[17:42:44][C][template.switch:058]: Template Switch 'TagReader LED enabled'
[17:42:44][C][template.switch:058]:   Icon: 'mdi:alarm-light-outline'
[17:42:44][C][template.switch:059]:   Restore State: NO
[17:42:44][C][template.switch:060]:   Optimistic: YES
[17:42:44][C][esp8266_pwm:022]: ESP8266 PWM:
[17:42:44][C][esp8266_pwm:023]:   Pin: GPIO13 (Mode: OUTPUT)
[17:42:44][C][esp8266_pwm:024]:   Frequency: 988.0 Hz
[17:42:44][C][fastled:019]: FastLED light:
[17:42:44][C][fastled:020]:   Num LEDs: 1
[17:42:44][C][fastled:021]:   Max refresh rate: 400
[17:42:44][C][logger:185]: Logger:
[17:42:44][C][logger:186]:   Level: DEBUG
[17:42:44][C][logger:187]:   Log Baud Rate: 115200
[17:42:44][C][logger:188]:   Hardware UART: UART0
[17:42:44][C][light:178]: Light 'TagReader LED'
[17:42:44][C][light:180]:   Default Transition Length: 0.0s
[17:42:44][C][light:181]:   Gamma Correct: 2.80
[17:42:44][C][pn532:321]: PN532:
[17:42:44][C][pn532:333]:   Update Interval: 1.0s
[17:42:44][C][pn532_i2c:041]:   Address: 0x24
[17:42:44][C][status:034]: Status Binary Sensor 'TagReader Status'
[17:42:44][C][status:034]:   Device Class: 'connectivity'
[17:42:44][C][rtttl:017]: Rtttl
[17:42:44][C][captive_portal:169]: Captive Portal:
[17:42:44][C][ota:029]: Over-The-Air Updates:
[17:42:44][C][ota:030]:   Address: tagreader.local:8266
[17:42:44][C][api:095]: API Server:
[17:42:44][C][api:096]:   Address: tagreader.local:6053

dmeus, 2020-11-22:
Tried switching the -well- switches (given that I first set them as pictured in the readme), still nothing. They are now in this position: according to my understanding 1 is in the up position now?

[image]

When I try to access the logs in esphome-dev, the logger shows nothing after ‘INFO Successfully connected to tagreader.local’. I’m guessing this is a red herring...

adonno, 2020-11-22:
Would you mind checking your solders?  SDA seems a bit "bad"  
Could you please also post a picture from your D1 as well? 

In addition, maybe try checking directly with the ESP connected to your PC and getting the logs via USB by using ESPHome flasher  https://github.com/esphome/esphome-flasher/releases  
Connect the ESP to your pc and select the right com port and hit view logs

ollimagic, 2020-11-23:
hi i tried everything different boards different cable but nothing how can
i log if something happen on the reader
in esplog i see that
[0;36m[D][wifi:339]: Found networks: [0m
[0;32m[I][wifi:385]: - 'FritzBox' [5m(2C:91:AB:96:80:23) [6m
[0;33mÔûéÔûäÔûå [0;37mÔûê [0m [0m
[0;36m[D][wifi:386]:     Channel: 1 [0m
[0;36m[D][wifi:387]:     RSSI: -59 dB [0m
[0;32m[I][wifi:385]: - 'FritzBox' [5m(5C:49:79:FE:E4:DB) [6m [0;33mÔûéÔûä
[0;37mÔûåÔûê [0m [0m
[0;36m[D][wifi:386]:     Channel: 6 [0m
[0;36m[D][wifi:387]:     RSSI: -71 dB [0m
[0;32m[I][wifi:385]: - 'FritzBox' [5m(98:9B:CB:56:73:6D) [6m [0;33mÔûéÔûä
[0;37mÔûåÔûê [0m [0m
[0;36m[D][wifi:386]:     Channel: 6 [0m
[0;36m[D][wifi:387]:     RSSI: -73 dB [0m
[0;36m[D][wifi:389]: - [5m'1xxBox' [6m [5m(44:4E:6D:28:0E:19) [6m
[0;33mÔûéÔûäÔûå [0;37mÔûê [0m [0m
[0;36m[D][wifi:389]: - [5m'xxZugang' [6m [5m(2E:91:AB:96:80:23) [6m
[0;33mÔûéÔûäÔûå [0;37mÔûê [0m [0m
[0;36m[D][wifi:389]: - [5m'' [6m [5m(B6:7C:9C:DC:A5:9A) [6m [0;33mÔûéÔûä
[0;37mÔûåÔûê [0m [0m
[0;36m[D][wifi:389]: - [5m'MieterZugang' [6m [5m(5E:49:79:FE:E4:DB) [6m
[0;33mÔûéÔûä [0;37mÔûåÔûê [0m [0m
[0;36m[D][wifi:389]: - [5m'FritzMa' [6m [5m(E8:94:F6:59:2C:01) [6m
[0;33mÔûéÔûä [0;37mÔûåÔûê [0m [0m
[0;36m[D][wifi:389]: - [5m'FritzMa' [6m [5m(E8:94:F6:59:38:B2) [6m
[0;33mÔûéÔûä [0;37mÔûåÔûê [0m [0m
[0;36m[D][wifi:389]: - [5m'FritzMa' [6m [5m(38:10:D5:36:59:C5) [6m
[0;33mÔûéÔûä [0;37mÔûåÔûê [0m [0m
[0;36m[D][wifi:389]: - [5m'FritzMa' [6m [5m(F4:F2:6D:F5:E3:1D) [6m
[0;31mÔûé [0;37mÔûäÔûåÔûê [0m [0m
[0;32m[I][wifi:194]: WiFi Connecting to 'FritzBox'... [0m
[0;32m[I][wifi:457]: WiFi Connected! [0m
[0;35m[C][wifi:303]:   SSID: [5m'FritzBox' [6m [0m
[0;35m[C][wifi:304]:   IP Address: 192.168.178.83 [0m
[0;35m[C][wifi:306]:   BSSID: [5m2C:91:AB:96:80:23 [6m [0m
[0;35m[C][wifi:307]:   Hostname: 'tagreader' [0m
[0;35m[C][wifi:311]:   Signal strength: -51 dB [0;33mÔûéÔûäÔûå [0;37mÔûê
[0m [0m
[0;35m[C][wifi:315]:   Channel: 1 [0m
[0;35m[C][wifi:316]:   Subnet: 255.255.255.0 [0m
[0;35m[C][wifi:317]:   Gateway: 192.168.178.1 [0m
[0;35m[C][wifi:318]:   DNS1: 192.168.178.1 [0m
[0;35m[C][wifi:319]:   DNS2: (IP unset) [0m
[0;36m[D][wifi:466]: Disabling AP... [0m
[0;35m[C][ota:029]: Over-The-Air Updates: [0m
[0;35m[C][ota:030]:   Address: tagreader.local:8266 [0m
[0;35m[C][api:022]: Setting up Home Assistant API server... [0m
[0;32m[I][app:059]: setup() finished successfully! [0m
[0;32m[I][app:105]: ESPHome version 1.16.0-dev compiled on Nov 22 2020,
16:28:28 [0m
[0;35m[C][wifi:443]: WiFi: [0m
[0;35m[C][wifi:303]:   SSID: [5m'FritzBox' [6m [0m
[0;35m[C][wifi:304]:   IP Address: 192.168.178.83 [0m
[0;35m[C][wifi:306]:   BSSID: [5m2C:91:AB:96:80:23 [6m [0m
[0;35m[C][wifi:307]:   Hostname: 'tagreader' [0m
[0;35m[C][wifi:311]:   Signal strength: -52 dB [0;33mÔûéÔûäÔûå [0;37mÔûê
[0m [0m
[0;35m[C][wifi:315]:   Channel: 1 [0m
[0;35m[C][wifi:316]:   Subnet: 255.255.255.0 [0m
[0;35m[C][wifi:317]:   Gateway: 192.168.178.1 [0m
[0;35m[C][wifi:318]:   DNS1: 192.168.178.1 [0m
[0;35m[C][wifi:319]:   DNS2: (IP unset) [0m
[0;35m[C][i2c:028]: I2C Bus: [0m
[0;35m[C][i2c:029]:   SDA Pin: GPIO4 [0m
[0;35m[C][i2c:030]:   SCL Pin: GPIO5 [0m
[0;35m[C][i2c:031]:   Frequency: 400000 Hz [0m
[0;35m[C][template.switch:058]: Template Switch 'TagReader Buzzer Enabled'
[0m
[0;35m[C][template.switch:058]:   Icon: 'mdi:volume-high' [0m
[0;35m[C][template.switch:059]:   Restore State: NO [0m
[0;35m[C][template.switch:060]:   Optimistic: YES [0m
[0;35m[C][template.switch:058]: Template Switch 'TagReader LED enabled' [0m
[0;35m[C][template.switch:058]:   Icon: 'mdi:alarm-light-outline' [0m
[0;35m[C][template.switch:059]:   Restore State: NO [0m
[0;35m[C][template.switch:060]:   Optimistic: YES [0m
[0;35m[C][esp8266_pwm:022]: ESP8266 PWM: [0m
[0;35m[C][esp8266_pwm:023]:   Pin: GPIO13 (Mode: OUTPUT) [0m
[0;35m[C][esp8266_pwm:024]:   Frequency: 1000.0 Hz [0m
[0;35m[C][fastled:019]: FastLED light: [0m
[0;35m[C][fastled:020]:   Num LEDs: 1 [0m
[0;35m[C][fastled:021]:   Max refresh rate: 400 [0m
[0;35m[C][logger:185]: Logger: [0m
[0;35m[C][logger:186]:   Level: DEBUG [0m
[0;35m[C][logger:187]:   Log Baud Rate: 115200 [0m
[0;35m[C][logger:188]:   Hardware UART: UART0 [0m
[0;35m[C][light:178]: Light 'TagReader LED' [0m
[0;35m[C][light:180]:   Default Transition Length: 0.0s [0m
[0;35m[C][light:181]:   Gamma Correct: 2.80 [0m
[0;35m[C][pn532:321]: PN532: [0m
[0;35m[C][pn532:333]:   Update Interval: 1.0s [0m
[0;35m[C][pn532_i2c:041]:   Address: 0x24 [0m
[0;35m[C][status:034]: Status Binary Sensor 'TagReader Status' [0m
[0;35m[C][status:034]:   Device Class: 'connectivity' [0m
[0;35m[C][rtttl:017]: Rtttl [0m
[0;35m[C][captive_portal:169]: Captive Portal: [0m
[0;35m[C][ota:029]: Over-The-Air Updates: [0m
[0;35m[C][ota:030]:   Address: tagreader.local:8266 [0m
[0;35m[C][api:095]: API Server: [0m
[0;35m[C][api:096]:   Address: tagreader.local:6053 [0m
[0;36m[D][main:119]: API is connected! [0m
[0;36m[D][rtttl:037]: Playing song success [0m
[0;36m[D][light:265]: 'TagReader LED' Setting: [0m
[0;36m[D][light:274]:   State: ON [0m
[0;36m[D][light:278]:   Brightness: 100% [0m
[0;36m[D][light:287]:   Red=0%, Green=0%, Blue=100% [0m
[0;36m[D][light:297]:   Flash Length: 0.5s [0m
[0;36m[D][binary_sensor:036]: 'TagReader Status': Sending state ON [0m
[0;36m[D][api.connection:604]: Client 'Home Assistant 0.118.2
(192.168.178.200)' connected successfully! [0m
[0;36m[D][rtttl:093]: Playback finished [0m
[0;32m[I][ota:046]: Boot seems successful, resetting boot loop counter. [0m

adonno, 2020-11-23:
with the ESP connect to the PC and logging, hit the reset button so the logs will be shown from the beginning 

As already asked before, would you mind sending a picture of your Wiring? The more picture the better, top, bottom, wiring, ...  etc.

ollimagic, 2020-11-23:
;l l£ƒ| älÓ|─dõc|Å├õô{█cäcä¹g'ƒl'o£Òýc8äçl;l{$p¹gÓâdî£bgÒ|ýdçÅcî¾og´ $îÄ$`É'od`n{çôøo#d`søÆ'cl £#çølsø`³├g£[0;32m[I][logger:166]: Log initialized[0m
[0;35m[C][ota:366]: There have been 0 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...[0m
[0;35m[C][fastled:010]: Setting up FastLED light...[0m
[0;35m[C][light:097]: Setting up light 'TagReader LED'...[0m
[0;36m[D][light:265]: 'TagReader LED' Setting:[0m
[0;36m[D][light:278]:   Brightness: 100%[0m
[0;36m[D][light:287]:   Red=100%, Green=100%, Blue=100%[0m
[0;35m[C][pn532:027]: Setting up PN532...[0m
[0;33m[W][i2c:070]: Received NACK on transmit of address 0x24[0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[0;33m[W][i2c:086]: Requesting 1 bytes from 0x24 failed![0m
[1;31m[E][pn532:031]: Error sending version command[0m
[1;31m[E][component:092]: Component was marked as failed.[0m
[0;36m[D][binary_sensor:034]: 'TagReader Status': Sending initial state OFF[0m
[0;35m[C][wifi:033]: Setting up WiFi...[0m
[0;36m[D][wifi:324]: Starting scan...[0m
[0;36m[D][wifi:339]: Found networks:[0m
[0;32m[I][wifi:385]: - 'FritzBox' [5m(2C:91:AB:96:80:23) [6m[0;32mÔûéÔûäÔûåÔûê[0m[0m
[0;36m[D][wifi:386]:     Channel: 1[0m
[0;36m[D][wifi:387]:     RSSI: -45 dB[0m
[0;32m[I][wifi:385]: - 'FritzBox' [5m(5C:49:79:FE:E4:DB) [6m[0;33mÔûéÔûä[0;37mÔûåÔûê[0m[0m
[0;36m[D][wifi:386]:     Channel: 6[0m
[0;36m[D][wifi:387]:     RSSI: -66 dB[0m
[0;32m[I][wifi:385]: - 'FritzBox' [5m(98:9B:CB:56:73:6D) [6m[0;33mÔûéÔûä[0;37mÔûåÔûê[0m[0m
[0;36m[D][wifi:386]:     Channel: 6[0m
[0;36m[D][wifi:387]:     RSSI: -78 dB[0m
[0;36m[D][wifi:389]: - [5m'MieterZugang'[6m [5m(2E:91:AB:96:80:23) [6m[0;32mÔûéÔûäÔûåÔûê[0m[0m
[0;36m[D][wifi:389]: - [5m'1und1Box'[6m [5m(44:4E:6D:28:0E:19) [6m[0;33mÔûéÔûäÔûå[0;37mÔûê[0m[0m
[0;36m[D][wifi:389]: - [5m''[6m [5m(B6:7C:9C:DC:A5:9A) [6m[0;33mÔûéÔûäÔûå[0;37mÔûê[0m[0m
[0;36m[D][wifi:389]: - [5m'MieterZugang'[6m [5m(5E:49:79:FE:E4:DB) [6m[0;33mÔûéÔûäÔûå[0;37mÔûê[0m[0m
[0;36m[D][wifi:389]: - [5m'FritzMa'[6m [5m(E8:94:F6:59:2C:01) [6m[0;33mÔûéÔûä[0;37mÔûåÔûê[0m[0m
[0;36m[D][wifi:389]: - [5m'FritzMa'[6m [5m(E8:94:F6:59:38:B2) [6m[0;31mÔûé[0;37mÔûäÔûåÔûê[0m[0m
[0;32m[I][wifi:194]: WiFi Connecting to 'FritzBox'...[0m
[0;32m[I][wifi:457]: WiFi Connected![0m
[0;35m[C][wifi:303]:   SSID: [5m'FritzBox'[6m[0m
[0;35m[C][wifi:304]:   IP Address: 192.168.178.83[0m
[0;35m[C][wifi:306]:   BSSID: [5m2C:91:AB:96:80:23[6m[0m
[0;35m[C][wifi:307]:   Hostname: 'tagreader'[0m
[0;35m[C][wifi:311]:   Signal strength: -53 dB [0;33mÔûéÔûäÔûå[0;37mÔûê[0m[0m
[0;35m[C][wifi:315]:   Channel: 1[0m
[0;35m[C][wifi:316]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:317]:   Gateway: 192.168.178.1[0m
[0;35m[C][wifi:318]:   DNS1: 192.168.178.1[0m
[0;35m[C][wifi:319]:   DNS2: (IP unset)[0m
[0;36m[D][wifi:466]: Disabling AP...[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: tagreader.local:8266[0m
[0;35m[C][api:022]: Setting up Home Assistant API server...[0m
[0;32m[I][app:059]: setup() finished successfully![0m
[0;32m[I][app:105]: ESPHome version 1.16.0-dev compiled on Nov 22 2020, 16:28:28[0m
[... 32 lines cut ...]
[0;35m[C][logger:188]:   Hardware UART: UART0[0m
[0;35m[C][light:178]: Light 'TagReader LED'[0m
[0;35m[C][light:180]:   Default Transition Length: 0.0s[0m
[0;35m[C][light:181]:   Gamma Correct: 2.80[0m
[0;35m[C][pn532:321]: PN532:[0m
[0;35m[C][pn532:333]:   Update Interval: 1.0s[0m
[0;35m[C][pn532_i2c:041]:   Address: 0x24[0m
[0;35m[C][status:034]: Status Binary Sensor 'TagReader Status'[0m
[0;35m[C][status:034]:   Device Class: 'connectivity'[0m
[0;35m[C][rtttl:017]: Rtttl[0m
[0;35m[C][captive_portal:169]: Captive Portal:[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: tagreader.local:8266[0m
[0;35m[C][api:095]: API Server:[0m
[0;35m[C][api:096]:   Address: tagreader.local:6053[0m
[0;36m[D][main:119]: API is connected![0m
[0;36m[D][rtttl:037]: Playing song success[0m
[0;36m[D][light:265]: 'TagReader LED' Setting:[0m
[0;36m[D][light:274]:   State: ON[0m
[0;36m[D][light:278]:   Brightness: 100%[0m
[0;36m[D][light:287]:   Red=0%, Green=0%, Blue=100%[0m
[0;36m[D][light:297]:   Flash Length: 0.5s[0m
[0;36m[D][binary_sensor:036]: 'TagReader Status': Sending state ON[0m
[0;36m[D][api.connection:604]: Client 'Home Assistant 0.118.2 (192.168.178.200)' connected successfully![0m
[0;36m[D][rtttl:093]: Playback finished[0m

ollimagic, 2020-11-23:
[image]

ollimagic, 2020-11-23:
[image]
[image]

ollimagic, 2020-11-23:
the connectors are all working i testet tham

ollimagic, 2020-11-23:
i mean the solder points have all connection sorry for my english

adonno, 2020-11-23:
[image]

Please invert the switchtes and retry

ollimagic, 2020-11-23:
now ist is working i change the nfc reade board again hurraaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa

ollimagic, 2020-11-23:
sorry for wasting your time i guess my boad was defect

closed 2020-11-23 by ollimagic

dmeus, 2020-11-23:
> Would you mind checking your solders? SDA seems a bit "bad"
> Could you please also post a picture from your D1 as well?
>
> In addition, maybe try checking directly with the ESP connected to your PC and [...]

I have double checked the solders and they are transferring a signal between the D1 and PN532. This works for both SDA and SCL. Photo of the D1 below. When I hook the device up to my laptop, open the logs in ESPHome Flasher and hold an NFC card against the reader it recognizes the tags.

[19:39:43][D][pn532:159]: Found new tag '04-2B-5C-EA-B3-6D-80'
[19:39:49][D][pn532:159]: Found new tag '04-2C-5F-EA-B3-6D-80'

[image]

dmeus, 2020-11-24:
OK, I deleted the node in ESPHome and compiled a new firmware. Flashed it to the D1, filled in wifi details via the captive portal and rebooted the whole thing. Same scenario as above. ESPHome shows no logs for the node except 'connected', ESPHome flasher shows everything as stated above when I open the node in HA. Tags are recognized by ESPHome Flasher when I present them but they trigger no automation. I can't for the life of me figure out where this is going wrong. Any more tips @adonno ?

adonno, 2020-11-24:
> OK, I deleted the node in ESPHome and compiled a new firmware. Flashed it to the D1, filled in wifi details via the captive portal and rebooted the whole thing. Same scenario as above. ESPHome shows [...]

Wait, hold on, the reader send the tag ID to HA? 
I thought that was the issue ? Well then it's an issue with your automation please share your automation .