adonno/tagreader #76: RFID Card and NFC Phone | Unconsistent behaviour

CarpeDiemRo, 2020-11-22:
**Prerequisites**
a. Wemos D1 mini
b. PN RFID 532
c. Card & tag, 13.56 MHz
d. master branch pn532_i2c

**Use case** 
I want to use my Huawei Mate 20 Pro as an NFC Tag

**What I did**
1. Open Home Assistant Companion which is logged in at HA instance
2. NFC Tag option from Settings
3. Write Tag on White Card
4. Returning results
```
- platform: event
  event_type: tag_scanned
  event_data:
    device_id: 93d872ca49dfe4d7
    tag_id: ce1685b2-0bdf-4786-bea4-39caa49c4035
```
5. Fire event button
6. Check if *ce1685b2-0bdf-4786-bea4-39caa49c4035* is displayed in Tags from configuration
[image]

7. Make the following automation in automation.yaml

```
- alias: "Huawei NFC Access"
  initial_state: True
  mode: single
  trigger:
  - platform: event
    event_type: tag_scanned
    event_data:
      device_id: 93d872ca49dfe4d7
      tag_id: ce1685b2-0bdf-4786-bea4-39caa49c4035
  action:
    - service: switch.toggle
      entity_id: switch.shelly_1pm
```
8. Reload automation
9. Swipe Huawei Phone

**Expected results**
1. The bulb must be switched on

**Actual results**
1. The bulb is not light on and a new entry is created in Tags
2. Huawei and White Card still have different identificators
[image]

**More actions**
1. Go To events and listen *tag_scanned*
2. Scan Huawei and White Card
--> "device_id": `bf45cac435a56ede5256ea83fed4f9c0` is the same for both.
--> tag id is different than the id created with HA Companion
--> device_id is different than the id created with HA Companion
[image]
--
3. Go to automation and at device_id paste `bf45cac435a56ede5256ea83fed4f9c0` value

*New automation code excluded tag_id*
```
- alias: "Huawei NFC Access Device ID"
  initial_state: True
  mode: single
  trigger:
  - platform: event
    event_type: tag_scanned
    event_data:
      device_id: bf45cac435a56ede5256ea83fed4f9c0
    #   tag_id: ce1685b2-0bdf-4786-bea4-39caa49c4035
  action:
    - service: switch.toggle
      entity_id: switch.shelly_1pm
```
4. Reload automation
5. Swipe White Card --> the bulb is switching ON
6. Swipe Huawei --> the bulb is switching ON
7. Check Tags from HA
New Tag is created for Huawei 
8. Swipe Huawei again
New Tag is created for Huawei
[image]

**Expected results**
Huawei and White card should be considered one entity
When I write a new card 

```
      device_id: 93d872ca49dfe4d7
      tag_id: ce1685b2-0bdf-4786-bea4-39caa49c4035
```
must be used in automation in order to make two entities to act as one

--
**More Testing**
1. Go To HA companion and read White Card
--> results

```
- platform: event
  event_type: tag_scanned
  event_data:
    device_id: 93d872ca49dfe4d7
    tag_id: ce1685b2-0bdf-4786-bea4-39caa49c4035
```
2. Make the following automation

```
- alias: "Huawei NFC Access"
  initial_state: True
  mode: single
  trigger:
  - platform: event
    event_type: tag_scanned
    event_data:
      device_id: 93d872ca49dfe4d7
      tag_id: ce1685b2-0bdf-4786-bea4-39caa49c4035
  action:
    - service: switch.toggle
      entity_id: switch.shelly_1pm
```
3. Fire the event directly from the HA companion / Button **Fire Event**
--> The bulb is switching ON and OFF
--> Verify-in Tags if `ce1685b2-0bdf-4786-bea4-39caa49c4035` was used == yes
4. Swipe Huawei in order to obtain the same result
--> a new tag is created
--> the bulb remains switched off

**In conclusion**
if I used *Fire Event* button directly from HA companion the phone trigger the action and in TAGS is used the device id which was created, but if I put the **same** values in the automation I am not able to obtain the same behavior

insajd, 2020-11-25:
will be fixed with https://github.com/adonno/tagreader/pull/52 merge

insajd, 2020-11-25:
this ticket related to https://github.com/adonno/tagreader/issues/41

jesserockz, 2020-12-01:
@CarpeDiemRo I think you are misunderstanding how the tags and the tagreader works. I will try to explain.

Your `White card` can be written with the mobile app to have the contents ID of `ce1685b2-0bdf-4786-bea4-39caa49c4035` but the tagreader cannot read this ID until #52 is merged and can only see the UID (set at the factory when the card is made) `35-C7-97-2A`

As for your `Huawei` while you can scan it using the tagreader, Android will **ALWAYS** return a random UID for every scan for security reasons. 

The device_id (`bf45cac435a56ede5256ea83fed4f9c0`) is an internal random ID given to any ESPHome device by HA when it is first added as an integration.

I hope this helps. 
Jesse

CarpeDiemRo, 2020-12-07:
@jesserockz Since what I said will be fixed with #52 merge, I think I understanding is quite good. 
@insajd Thanks for your contribution and for your work. When do you estimate that the merge branch will be made public in order to use it?

jesserockz, 2020-12-07:
@CarpeDiemRo Are you trying to use the `tagreader` to scan the `white card` **and** scan your `huawei`?
Or are you trying to scan the `white card` with the `tagreader` and scan the `white card` with your `huawei`?

insajd, 2020-12-07:
@CarpeDiemRo sorry I have not made contributions or work to this component. I might have misunderstood what you meant from my perspective of what I tried to achieve, and actually did it with help of @jesserockz  👍

CarpeDiemRo, 2020-12-07:
@insajd Got it. In conclusion, this is my use case: I want to use my phone in a similar way as I use the white card. For this, helping by HA companion for Android I write UUID on the white card. But every time when I use the phone it generates a different tag number -> which is normal. My expectations were to have a static tag ID on my phone and on my white card. 
As far as I understood this is not possible. My question is: _There are any possibilities to use my phone?_

jesserockz, 2020-12-07:
Yeah thats what I got from your initial message. No this will not be possible as its built into the Android system to generate a random UID every scan.
A "workaround" could be to get one of the sticker tags and put it somewhere on your phone where the antenna is not and scan that with the tagreader.

CarpeDiemRo, 2020-12-07:
Oki Doki. Clear. One more question. Do you know what could be the purpose of the **Tag Write** option from HA companion?

jesserockz, 2020-12-07:
Yes, that writes the UUID (`ce1685b2-0bdf-4786-bea4-39caa49c4035` as a URL) to the card as NDEF data so that a scan of the card from a phone will auto launch the HA app and send the event.

CarpeDiemRo, 2020-12-08:
@jesserockz Understood. Thank you.

FrancYescO, 2021-04-13:
if for (Android only?) phones there is no way to have a fixed UID, is there a way we can get it from contactless credit cards? for example, generate one using the card PAN?
i've tested 4 CC:
1 VISA + 1 MC + My Xiaomi MI9: random UID each scan (like reported above)
1 MC: same UID each scan
1 MC: same 12 character (instead of 6) UID each scan

danielbrunt57, 2021-07-15:
I have successfully scanned my Driver's license , 4 Mastercards, RBC VISA, and my TD Bank client card.  My Scotiabank VISA card and RBC client cards do not scan...

closed 2021-12-25 by adonno