adonno/tagreader #78: Created a Remix of your case Cover

dudu631, 2020-11-26:
Hi, I remixed one cover of your 3D model cases. 

One that has thinner space for the LED to show up better when using darker colors (like black icon). And one that has a thin grid to make the buzzer a little bit louder for those who need it. 

The link is [here](https://www.thingiverse.com/thing:4666280).

Thanks!

iz3man, 2020-11-26:
Like it! Nice. I plan to use the reader for ALARM SYSTEM de/activation. So I will use different colors for the LED as well. A thinner LED "cover" is very welcome.

dudu631, 2020-11-26:
Actually not a lot. It's small distances, most printers should be good. 

Mine is a prusa mk3s, so no issues printing that way.

closed 2021-09-17 by adonno