adonno/tagreader #81: Mqtt instead of homeassistant API

malesci, 2020-12-02:
Thank you for this project.
Is It possibile use mqtt instead of homeassistant API?
Can you provide an example in case?
Thank you

mpmoeller, 2020-12-09:
Hi @malesci,
I just started putting the parts together, but looking at the documentation it seems possible, see: https://esphome.io/components/mqtt.html

Still, the ESPHome folks have some points why to prefer the native API, see:
https://esphome.io/components/api.html (close to the end)

Good luck,
Martin

iz3man, 2021-01-05:
I'm not sure if it's worth opening a new thread for this, but how would I control the LED and BUZZER?
I want to use this for an alarm system, so the LED should be GREEN when disarmed, acknowledge the read of NFC tag and flash blue and BEEP, and then turn the LED red blinking when ARMED, and BEEP every some seconds.
How could that be acchieved. 
Sorry. I didn't find and info about this and where to look further. No idea about the HA API either...
Thanks

Noschvie, 2021-07-08:
@mpmoeller : martin, did you get MQTT running ?
Maybe you can share your config ?
Thanks!

mpmoeller, 2021-07-08:
@Noschvie - nope. I accidentally destroyed my reader, so I didn't look into that topic for a while.

Noschvie, 2021-07-08:
Has anybody a valid MQTT client config to send the scanned value ?
Thanks!

adonno, 2021-09-17:
wanted to look into it but haven't had time yet will do so as soon as I have some time on my hands

adonno, 2022-04-11:
Can you try this one ? 

https://github.com/adonno/tagreader/pull/164/files