adonno/tagreader #83: Put PN532 Switches in Images + Schematic to I²C

mpmoeller, 2020-12-09:
Hey,

Possibly like many others, the _recent_ Home Assistant announcements brought me to your project. I literally just finished my first self-soldered tag reader 🥳 . Clearly lacking talent, the first device ended up in smoke, the second one had a buzzer for 2 seconds only before gravity decided that it will do well enough w/o it 🙈.

Disclosing that I am a total beginner I have a small feedback on documentation:

**_Please put the PN532 switches in your images + schematic to `I²C`._**

AFAICT, the pictures used in the [README.md](https://github.com/adonno/tagreader/blob/master/README.md) still have the switches put to `SPI`, see:

- https://github.com/adonno/tagreader/blob/master/docs/cases.jpg
- https://github.com/adonno/tagreader/blob/master/docs/open-case.jpg

In the schematic, the switches are set to `HSU`, see:

- https://github.com/adonno/tagreader/blob/master/Schematics/tag_reader_schematics_v2.png

Of course, I read that the switches have to be put to `I²C` and afterwards I learned that this was introduced with release 4.0. But being new in that game, things easily overwhelm and I was unsure if the positions were correct. So any (visual) guidance is helpful.

In my particular case, PN532 wasn't scanning because I simply inverted the setting. Luckily only 2 options exist, hence I was able to solve the "problem" myself. But for future beginners to come, you might want to spare them efforts and headaches.

Thanks,
Martin

waazdakka, 2021-01-07:
Hello mpmoeller!
What setting are you talking about, the switches ?
I followed the schematics (https://github.com/adonno/tagreader/blob/master/Schematics/tag_reader_schematics_v2.png) and not the photo of the finished case, but I never succeed to have neither the scanner working or the led on :(
I can't figure out what I've done wrong on here!
A little help ? 
Thanks!!!

mpmoeller, 2021-01-07:
Hey @waazdakka,

In the schematics you see the big red PN532 board at the right.

On board you see a little black box w/ yellow switches and a „legend“ how to put them.

The schematics show them as 0/0 (HSU), but to make the PN532 run w/ I2C you need to put them to 1/0, i.e. change one switch.

Orherwise you won‘t hear any beeps or se any LED flashing as the PN532 won‘t „talk“ to the main processor.

The change to I2C only came recently. It is mentioned in the text, but I suspect that the pictures show the state „before“.

Best.

waazdakka, 2021-01-08:
Hey,

Thanks for your fast answer!

I followed the Readme instructions and your indications, but I don't have any success with my tagreader 😭
Major question : the schematic is completeley different than the open case photo!

While the schematics use SDA and SCL ports, the photo shows wires on MSO, MOSI and SS ports.
On the Wifi board, the D3 port is solder on the photo, while it's not in use on the schematic 🙄

I followed the schematic with no success, maybe I should follow the photo?

Thanks again!

mpmoeller, 2021-01-08:
I followed the schematic and surrounding text w/o any problems (except the switches) 

Good luck!

Re4DeR, 2021-01-20:
thanks. after inverting switch setting, it works

adonno, 2021-05-01:
update the images

closed 2021-05-01 by adonno