adonno/tagreader #84: STEP Files

stloewen, 2020-12-12:
Hi could you share the step files in addition to the STLs? I'd like to modify the case to fit the RC522 reader.

adonno, 2021-01-05:
Hello, 
the Step files are a mess I could share them privately but since it was my first project I'm not feeling like sharing them not because they are confidential but because it's a mess. and every change might break them

lnxbil, 2021-01-30:
Hi,

I also wanted to change the design of the box. I have a lot of V3 Wemos D1 lying around which have the USB-Port on the other side, so that it does not fit. Redesigning everything from scratch is possible, but very time-consuming.

Best,
Andreas

stloewen, 2021-02-14:
This is my design for a D1 mini and a RC522 reader:

STEP-File: [TagReader_RC522_D1mini_Case.step.zip](https://github.com/adonno/tagreader/files/5978244/TagReader_RC522_D1mini_Case.step.zip)
OnShape-Link: https://cad.onshape.com/documents/5de3dc17f60bdd23e35afc2c/v/c33122f442e353dffec4cdcd/e/942aeed32624e93e8989da17
License: CC0

closed 2021-12-25 by adonno