adonno/tagreader #93: Errors when scanning tags

richardshaylor, 2021-01-14:
Hi,

I brought a tag reader pre-assembled from your store.

I've only just managed to get around to playing with this. 

It connected to homeassistant perfectly however the tag reader will only scan intermittently. When it does scan also the following error is displayed in the log:

[16:04:34][D][pn532:159]: Found new tag '83-F4-57-03'
[16:04:34][D][rtttl:093]: Playback finished
[16:04:35][W][i2c:070]: Received NACK on transmit of address 0x24
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:35][W][pn532:099]: Requesting tag read failed!

Any suggestions. I've checked the wiring and everything looks fine. Full log below.

Richard

[16:03:35][D][main:118]: API is connected!
[16:03:35][D][rtttl:037]: Playing song success
[16:03:35][D][light:264]: 'TagReader 28aa56 LED' Setting:
[16:03:35][D][light:273]:   State: ON
[16:03:35][D][light:277]:   Brightness: 100%
[16:03:35][D][light:286]:   Red=0%, Green=0%, Blue=100%
[16:03:35][D][light:289]:   White Value: 0%
[16:03:35][D][light:296]:   Flash Length: 0.5s
[16:03:35][D][api.connection:604]: Client 'Home Assistant 2021.1.1 (192.168.1.126)' connected successfully!
[16:03:35][D][rtttl:093]: Playback finished
[16:03:48]{d[I][logger:166]: Log initialized
[16:03:48][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[16:03:48][I][app:029]: Running through setup()...
[16:03:48][C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[16:03:48][C][fastled:010]: Setting up FastLED light...
[16:03:48][C][light:097]: Setting up light 'TagReader 28aa56 LED'...
[16:03:48][D][light:264]: 'TagReader 28aa56 LED' Setting:
[16:03:48][D][light:277]:   Brightness: 100%
[16:03:48][D][light:286]:   Red=100%, Green=100%, Blue=100%
[16:03:48][D][light:289]:   White Value: 0%
[16:03:48][C][pn532:027]: Setting up PN532...
[16:03:48][D][pn532:042]: Found chip PN532
[16:03:48][D][pn532:043]: Firmware ver. 1.6
[16:03:48][C][wifi:033]: Setting up WiFi...
[16:03:48][D][wifi:324]: Starting scan...
[16:03:55][D][wifi:339]: Found networks:
[16:03:55][I][wifi:385]: - 'SHAYLOR' (FC:EC:DA:8A:48:D7) ▂▄▆█
[16:03:55][D][wifi:386]:     Channel: 1
[16:03:55][D][wifi:387]:     RSSI: -49 dB
[16:03:55][D][wifi:389]: - '💩' (FE:EC:DA:8A:48:D7) ▂▄▆█
[16:03:55][D][wifi:389]: - '' (0E:EC:DA:8A:48:D7) ▂▄▆█
[16:03:55][D][wifi:389]: - 'VM7752782' (40:0D:10:E6:CC:29) ▂▄▆█
[... 81 lines cut ...]
[16:03:59][D][rtttl:093]: Playback finished
[16:04:00][D][pn532:159]: Found new tag '83-F4-57-03'
[16:04:02][W][i2c:070]: Received NACK on transmit of address 0x24
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:04:02][W][pn532:099]: Requesting tag read failed!

richardshaylor, 2021-01-14:
As a follow up looking at the logs when it stops working the log is just keeps repeating the below until i reset:

[16:23:02][W][i2c:070]: Received NACK on transmit of address 0x24
[16:23:04][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:23:04][W][pn532:099]: Requesting tag read failed!
[16:23:05][W][i2c:070]: Received NACK on transmit of address 0x24
[16:23:07][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:23:07][W][pn532:099]: Requesting tag read failed!
[16:23:08][W][i2c:070]: Received NACK on transmit of address 0x24

adonno, 2021-01-14:
Hey Richard 

Do you by any chance have a soldering iron ? 
There seems to be a contact issue on the pn532

I usally tested them all but maybe a wire came inexplicably loose during transit. They might look fin but wouldn‘t hurt hitting them with a soldering iron 

Please let me
Know how that went 

Best regards 
Andrea donni

richardshaylor, 2021-01-14:
Yes I do 😊

Thanks for your speedy response.

richardshaylor, 2021-01-14:
Must have been a loose connection :)

closed 2021-01-14 by richardshaylor