adonno/tagreader #98: Log message Timed out waiting for readiness from PN532!

D3rax, 2021-02-05:
Hi,

I have rewired my two tag readers  to the i2c configuration and updated ESPhome to the latest version.
Both tagreaders read tags but my logs show two weird messages:

1. Every second(!!) these two messages show up:
[14:46:14][V][pn532_i2c:029]: Timed out waiting for readiness from PN532!
[14:46:15][V][pn532:249]: Reading response

2. Every now and then I get the massage "[14:42:15][D][api:067]: Disconnecting Home Assistant 2021.2.0"

I'm using ESP8266 NodeMCU boards.

lnxbil, 2021-02-12:
I also see 1 on my system, but I think this is normal. You wait for a tag to be read and if nothing is detected, the call times out. Besides the debug messages, the reading works as expected.

a8ree, 2021-03-23:
I'm seeing the same error

```
[22:51:47][V][pn532:233]: Reading ACK...
[22:51:47][V][pn532:244]: ACK valid: YES
[22:51:47][V][pn532_i2c:040]: Reading response
[22:51:47][V][pn532_i2c:029]: Timed out waiting for readiness from PN532!
[22:51:47][V][pn532:255]: Turning RF field OFF
[22:51:47][V][pn532:233]: Reading ACK...
[22:51:47][V][pn532:244]: ACK valid: YES
[22:51:48][V][pn532:233]: Reading ACK...
[22:51:48][V][pn532:244]: ACK valid: YES
[22:51:48][V][pn532_i2c:040]: Reading response
[22:51:48][V][pn532_i2c:029]: Timed out waiting for readiness from PN532!
[22:51:48][V][pn532:255]: Turning

```
I'm trying to get this set up for the first time. I have enabled debugging and see this message but nothing different when the tag is placed in front.

adonno, 2021-05-01:
Mind sending a picture of your soldering?  
seems there is an issue

adonno, 2021-09-17:
no reply for at least 60 days

closed 2021-09-17 by adonno