adonno/tagreader #103: Unable to Flash

seanmccabe, 2021-03-20:
Hi there,

I am totally new to this and done my best to Google and try and figure this out.

I have wired everything up, and plyggin power into the board via USB both the boards light up, but I am having issues flashing this.

In ESPHome home in Docker I've copied in the tagreader.yaml and compiled. Plug the board in by USB and can see USB Serial.

Attempt to then upload over USB and get the following:

```
INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/tagreader/.pioenvs/tagreader/firmware.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 115200 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/tagreader/.pioenvs/tagreader/firmware.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
```

Really don't know what I am doing wrong here, and everything I Google just confuses me more, so looking for some plain english help.

Thanks!

zuidwijk, 2021-03-20:
Error message is easy: it can't connect.  

Either a faulty esp, or there is something connected to the rx or tx, or the usb chip is faulty. 

Either way... no connection between the esp and usb chip

seanmccabe, 2021-03-21:
Definately nothing in the rx or tx, completely untouched.
Will try flashing another unwired (didnt realise you could do this).

Thanks.

seanmccabe, 2021-03-21:
Ok, deffo faulty chip, or something else isnt quite right.
As just flashed an unsoldered unit without issue.

Maybe just to check Im doing this right (long time since I've done electronics).
The 5v line, is that one cable to the LED, and one to the PN532 yes? Both just soldered at the 5v terminal on the D1 mini.

zuidwijk, 2021-03-21:
Now I do remember that a faulty led could cause this too. In that case, unsolder the led. See if it works.

seanmccabe, 2021-03-28:
OK so second attempt at soldering, programmed beforehand, finished soldering and doesn't work.
Anyone see what I'm doing wrong, I'm not the best, but can't see any glaring issues...

[image]
[image]
[image]
[image][image][image][image][image]

seanmccabe, 2021-03-28:
> Now I do remember that a faulty led could cause this too. In that case, unsolder the led. See if it works.

Ok, turns out yes, it is the LED, am i soldering it wrong? Just to make sure before i return them as faulty.

Also, is there a way to solder it in the meantime without the LED?

seanmccabe, 2021-04-04:
Sorted

closed 2021-04-04 by seanmccabe