adonno/tagreader #105: pn532 not working

alesaund, 2021-04-12:
after upgrading esphome to 1.6 i had to upload the last yaml with i2c mode but fails with this log:
 
[16:22:09][C][pn532:015]: Setting up PN532...
[16:22:10][W][i2c:070]: Received NACK on transmit of address 0x24
[16:22:10][V][pn532:233]: Reading ACK...
[16:22:12][W][i2c:086]: Requesting 1 bytes from 0x24 failed!
[16:22:12][V][pn532_i2c:029]: Timed out waiting for readiness from PN532!
[16:22:12][E][pn532:019]: Error sending version command
[16:22:12][E][component:092]: Component was marked as failed.
[16:22:12][D][binary_sensor:034]: 'TagReader_3 Status': Sending initial state OFF

adonno, 2021-05-01:
Can you send a picture of your soldering?

alesaund, 2021-05-03:
[image]
[image]

alesaund, 2021-05-19:
wrong soldering. sorry my fault

closed 2021-05-19 by alesaund