adonno/tagreader #107: tag reader not detected

dominiquevocat, 2021-04-22:
i set up the private wifi and the device has an ip address but home assistant does not detect it. Running 2021.4.6

what can i do to try to get it to work?

zuidwijk, 2021-04-22:
Can you ping its name? <device name>.local

dominiquevocat, 2021-04-22:
yes

adonno, 2021-04-22:
That's because you aren't using default_config 

Go to you integrations, click on the orange plus sign , search for esphome 
Enter the ip address of the reader and voila

dominiquevocat, 2021-04-22:
ah, works.

dominiquevocat, 2021-04-22:
not sure what you mean by default config though

dominiquevocat, 2021-04-25:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/tag/__init__.py", line 112, in async_scan_tag
    raise HomeAssistantError("tag component has not been set up.")
homeassistant.exceptions.HomeAssistantError: tag component has not been set up.
```

So i guess by default config you mean configuration.yaml? I use home assistant since like version 0.8 or something so i frequently run into stuff not enabled :-) by default. Would you care to to let me know what seems to be missing in the config for the setup of the tag component?

adonno, 2021-04-25:
In configuration.yaml add

default_config:

https://www.home-assistant.io/integrations/default_config/

dominiquevocat, 2021-04-26:
In hindsight it was rather obvious but thank you!
Solved.

closed 2021-04-26 by dominiquevocat