adonno/tagreader #108: Flash fails with: Resolving IP address of tagreader.local

bsrdjan, 2021-04-25:
Hi,

just assembled the tag-reader and connected via usb to laptop running HA. The flash fails with 

`Program failed with code 1`

The red LED on NFC module is on and a quiet permanent tone can be heard from buzzer.

Status:

ESPHome: Materializev1.16.2 D
HA: core-2021.4.6

Log
```
INFO Reading configuration /config/esphome/tagreader.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/tagreader
Processing tagreader (board: d1_mini; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.8.5
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/tagreader/.pioenvs/tagreader/src/main.cpp.o
Linking /data/tagreader/.pioenvs/tagreader/firmware.elf
Retrieving maximum program size /data/tagreader/.pioenvs/tagreader/firmware.elf
Building /data/tagreader/.pioenvs/tagreader/firmware.bin
Checking size /data/tagreader/.pioenvs/tagreader/firmware.elf
RAM:   [=====     ]  48.8% (used 39976 bytes from 81920 bytes)
Flash: [=====     ]  45.7% (used 477776 bytes from 1044464 bytes)
Creating BIN file "/data/tagreader/.pioenvs/tagreader/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and "/data/tagreader/.pioenvs/tagreader/firmware.elf"
========================= [SUCCESS] Took 11.59 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of tagreader.local
ERROR Error resolving IP address of tagreader.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname
```

Any ideas what goes wrong?

adonno, 2021-04-25:
No need to flash it they came pre-flashed already 

if you want to flash an device the first time you will need to use esphome-flasher and connect the esp to your pc
esphome-flasher can also be used to check the logs 

https://github.com/esphome/esphome-flasher/releases/tag/v1.3.0

bsrdjan, 2021-04-26:
Thanks adonno. Tried with 1.3.0 and got following log:

```shell
Using '/dev/cu.usbserial-144440' as serial port.
Connecting........_____....._____....._____....._____....._____....._____....._____
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: 
Timed out waiting for packet header
```
Flashing another D1-mini w/o NFC module works fine. Does wiring looks correct?

[image]

adonno, 2021-05-01:
Seems to be a soldering issue there must be a short somewhere. Unfortunately, I can't troubleshoot and from what I see from the images, they look fine. But can be finicky sometimes.  Maybe try **carefully** desoldering the PN532. 

⚠️ The PN532 has very fragile connectors

bsrdjan, 2021-05-03:
> ⚠️ The PN532 has very fragile connectors

Thanks @adonno, re-soldering fixed the issue. Tags are detected only the buzzer and LED are not reacting when the tag is read. On power-up, the buzzer beeps and the LED flashes. 

The buzzer works from dev tools but no sound (and no LED) on tag reading. Tag shows up in home assistant
```yaml
service: esphome.tagreader1_play_rtttl
data:
  song_str: >-
    beep:d=16,o=5,b=100:b
```

adonno, 2021-05-13:
The need to be enabled within HA when adding tag reader you got 4 new entities

closed 2021-05-13 by adonno