adonno/tagreader #109: Tagreader wifi not showing up

dylanmcdermott, 2021-04-26:
Just got a tag readr and plugged it in via micro usb, and am not seeing the wifi network on either my iPhone or mac. Any troubleshooting tips?

adonno, 2021-05-01:
sorry for the late reply. 
Regarding your reader. First of all, I would like to apologize for the issues you are having 

I would like to try to fix the issue with you before having to send out a new one. 

Could you please download espHome flasher for your pc here: https://github.com/esphome/esphome-flasher/releases/tag/v1.3.0 

Start Esphome flasher
Then connect to Reader to your pc and hit "show logs" (could you try different cables as well, some USB cables only provide power and no data you can test it by connecting a device you are sure is working with USB to your pc an see if it communicates correctly) 

try to scan a card it should show you a message. 
if not, please try to hit the reset button via the "pinhole" on the side. 

Please let me know if the device shows anything in the logs 
Btw the reader will need 1minute to start the access point mode

pfefferle, 2021-05-10:
I have a very similar issue.

All LEDs seem to work fine, but I see no WLAN and I can also not see it in Esprom Flasher or in ESPHome (when I connect it to my HASS). Is there anything else I can try. Any easy way to check if USB is broken?

Thanks,
Matthias

pfefferle, 2021-05-10:
I see no logs on ESPHome and can not access the NFC reader at all.

adonno, 2021-05-13:
Hello @pfefferle  sorry for the late reply I've sent you an email 
@dylanmcdermott I didn't hear back from you, is your issue fixed?

dylanmcdermott, 2021-05-14:
HI @adonno no it never ended up working. I can try and reflash it.

adonno, 2021-06-15:
closing the issue sorted it out by more troubleshooting

closed 2021-06-15 by adonno