adonno/tagreader #110: NFC Card with two Tag IDs

chr55, 2021-04-27:
I have received my Adonno tag reader in the mail. I now have it integrated into Home Assistant. 

I have an NFC card that when I scan can give two different Tag Ids in Home Assistant. 

Sometimes when I scan the card it gives
Tag ID: 62537fed-6852-40fe-bffd-7c9ab8600dfc

Other times when I scan it the card send to Home Assistant
Tag ID: 04-F6-D9-F2-1F-68-80

It is only these two Tag IDs that are sent to Home Assistant. There is no consistency with the ID that is sent, it can be either and I am not able to predict.

Do I need to write my automations that can handle either Tag ID for this card or am I doing something wrong / missing some configuration somewhere?

Thank you!

adonno, 2021-05-01:
I'd suggest you go to your developer tools, and chose write_tag_id

[image]

jesserockz, 2021-05-01:
I think the issue here is that sometimes it can read the ndef uuid, and sometimes it fails, and falls back to the card uid

closed 2021-09-17 by adonno