adonno/tagreader #112: can not build tagreader from command line

lecrams, 2021-05-03:
Hi, When I build the tagreader.yaml from Hassio ESPHome everthing builds fine.
However when I try this from the command line I get:

esphome tagreader1.yaml run                                                                                                                      
INFO Reading configuration tagreader1.yaml
Failed config                                                                                                                                           
                                                                                                                                                        
pn532_i2c: [source tagreader1.yaml:131]                                                                                                                 
                                                                                                                                                        
  Component not found: pn532_i2c.

What do I need to do to get this building?

adonno, 2021-06-15:
@jesserockz is that a ESPHome issue ? 
it's wierd it can't find the component

jesserockz, 2021-06-15:
Could be esphome version, or a custom_component overriding with errors in python code.

What ESPHome version are you using?

adonno, 2021-09-17:
no reply since 60 days issue closed

closed 2021-09-17 by adonno