adonno/tagreader #120: Can't compile - error in example code?

zygote279, 2021-06-27:
Hi there,

Following this tutorial and trying to validate the example code you post here: https://github.com/adonno/tagreader/blob/master/tagreader.yaml

Not entirely sure what you mean by 'point esphome here', but have tried copying/pasting this code into my esphome instance (and updating the details, eg SSID/PWD) and it returns the following error:

`
[name_add_mac_suffix] is an invalid option for [esphome]. Please check the indentation.
  name_add_mac_suffix: True
  
  [project] is an invalid option for [esphome]. Please check the indentation.
`

Is there something daft I'm missing here?

adonno, 2021-07-02:
what version of esphome are you using?  make sure t have the latest

adonno, 2021-09-17:
no reply in 60 days, issue closed

closed 2021-09-17 by adonno