adonno/tagreader #121: Cannot read nfc cards.

willemdej, 2021-06-30:
I have programmed an nfc card, when I hold the nfc card above the tag reader. there will be no automation activated.
I have set the debugging. very verbose this is my log;

INFO Reading configuration /config/esphome/tagreader.yaml...
INFO Starting log output from 192.168 using esphome API
INFO Connecting to 192.168:6053 (192.168.)
INFO Successfully connected to 192.168
[17:39:16][I][app:105]: ESPHome version 1.19.4 compiled on Jun 26 2021, 19:50:31
[17:39:16][C][wifi:484]: WiFi:
[17:39:16][C][wifi:344]:   SSID: [redacted]
[17:39:16][C][wifi:345]:   IP Address: 
[17:39:16][C][wifi:347]:   BSSID: [redacted]
[17:39:16][C][wifi:348]:   Hostname: 'tagreader'
[17:39:16][C][wifi:352]:   Signal strength: -46 dB ▂▄▆█
[17:39:16][V][wifi:354]:   Priority: 0.0
[17:39:16][C][wifi:356]:   Channel: 6
[17:39:16][C][wifi:357]:   Subnet: 
[17:39:16][C][wifi:358]:   Gateway: 
[17:39:16][C][wifi:359]:   DNS1: 
[17:39:16][C][wifi:360]:   DNS2: (IP unset)
[17:39:16][C][i2c:028]: I2C Bus:
[17:39:16][C][i2c:029]:   SDA Pin: GPIO4
[17:39:16][C][i2c:030]:   SCL Pin: GPIO5
[17:39:16][C][i2c:031]:   Frequency: 400000 Hz
[17:39:16][C][template.switch:058]: Template Switch 'TagReader Buzzer Enabled'
[17:39:16][C][template.switch:058]:   Icon: 'mdi:volume-high'
[17:39:16][C][template.switch:059]:   Restore State: YES
[17:39:16][C][template.switch:060]:   Optimistic: YES
[17:39:16][C][template.switch:058]: Template Switch 'TagReader LED enabled'
[17:39:16][C][template.switch:058]:   Icon: 'mdi:alarm-light-outline'
[17:39:16][C][template.switch:059]:   Restore State: YES
[17:39:16][C][template.switch:060]:   Optimistic: YES
[17:39:16][C][esp8266_pwm:022]: ESP8266 PWM:
[17:39:16][C][esp8266_pwm:023]:   Pin: GPIO13 (Mode: OUTPUT)
[17:39:16][C][esp8266_pwm:024]:   Frequency: 988.0 Hz
[17:39:16][C][fastled:019]: FastLED light:
[17:39:16][C][fastled:020]:   Num LEDs: 1
[17:39:16][C][fastled:021]:   Max refresh rate: 400
[17:39:16][C][logger:189]: Logger:
[17:39:16][C][logger:190]:   Level: VERY_VERBOSE
[17:39:16][C][logger:191]:   Log Baud Rate: 115200
[17:39:16][C][logger:192]:   Hardware UART: UART0
[17:39:16][C][light:188]: Light 'TagReader LED'
[17:39:16][C][light:190]:   Default Transition Length: 0.0s
[17:39:16][C][light:191]:   Gamma Correct: 2.80
[17:39:16][C][pn532:344]: PN532:
[17:39:16][C][pn532:356]:   Update Interval: 1.0s
[17:39:16][C][pn532_i2c:125]:   Address: 0x24
[17:39:16][C][status:034]: Status Binary Sensor 'TagReader Status'
[17:39:16][C][status:034]:   Device Class: 'connectivity'
[17:39:16][C][rtttl:017]: Rtttl
[17:39:16][C][captive_portal:148]: Captive Portal:
[17:39:16][C][ota:029]: Over-The-Air Updates:
[17:39:16][C][ota:030]:   Address: 192.168.8266
[17:39:16][C][api:095]: API Server:
[17:39:16][C][api:096]:   Address: 192.168.:6053
[17:39:17][VV][scheduler:152]: Running interval '' with interval=5000 last_execution=5882826 (now=5887830)
[17:39:17][VV][light.addressable:168]: Addressable Light 'TagReader LED' (effect_active=NO next_show=NO)
[17:39:17][VV][light.addressable:172]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[17:39:17][VV][light.addressable:174]: 
[17:39:21][VV][api.service:220]: on_ping_request: PingRequest {}
[17:39:21][VV][api.service:032]: send_ping_response: PingResponse {}
[17:39:22][VV][scheduler:152]: Running interval '' with interval=5000 last_execution=5887826 (now=5892828)
[17:39:22][VV][light.addressable:168]: Addressable Light 'TagReader LED' (effect_active=NO next_show=NO)
[17:39:22][VV][light.addressable:172]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[17:39:22][VV][light.addressable:174]:

adonno, 2021-07-16:
Is that a purchased tag reader or one you assembled?  I would suggest connecting you reader to your PC via USB and using ESPHome Flasher to read into the logs, If there is any issue with the PN532 chip it will only show on the boot

adonno, 2021-09-17:
no reply  for 60 days , issue closed

closed 2021-09-17 by adonno