adonno/tagreader #124: Received 2 new card readers but 1st one is not working...

danielbrunt57, 2021-07-15:
I received my two tag readers yesterday, thank you.  The first one I was able to connect to its wi-fi URL and configured it to connect to my wi-fi and it showed up in HA but it will not scan any tags.  I moved onto the 2nd one and immediately noticed a difference between the two... the first one has no indicator lights after power up whereas the 2nd one lit up with a red LED.  The 1st one is successfully connected to my wi-fi as I can ping its DHCP address but nothing beyond that.  It does emit a sound shortly after being powered up but no red LED and no scans.  I can turn on the blue LED from HA and have noticed the blue LED is somewhat different than the 2nd reader... the first looks like four dots in a circle but the 2nd reader looks like a single bright blue LED.

danielbrunt57, 2021-07-15:
[image]
[image]

danielbrunt57, 2021-07-16:
Fixed!

Opened the case and found a bad solder joint on the ground...
[image]

Also found the reason for the difference in the blue LED's...
[image]

adonno, 2021-07-16:
Hey, @danielbrunt57 I am sorry for the poorly soldered GND I usually test them all and take great care in making sure I don't send defective devices. 

I also seem to have sent you TWO Different versions of the LID which is also not standard practice. 
Is there anything I can do?

danielbrunt57, 2021-07-16:
You can send me a new lid so they are the same.  I carefully dremelled the white plastic away on the 4 dot version so I see a blue circle from inside it now but it still appears as 4 dots which I can see from the back side if I hold it up to a light.

While I got the first one up and running after re-soldering the GND, the green LED is not working.  Blue is working and if I turn that on and then scan, it just turns off rather than changing to green like my 2nd reader does.  Could that be another bad solder joint?

nielsrune, 2021-07-21:
Just to add to this, my LED isn't working at all.
After seeing this post, I opened the lid which has the hole (equal to the bottom one in OP's picture)

But the LED won't turn on, even when switching on in HA :-(

adonno, 2021-08-10:
@danielbrunt57   @nielsrune  would you mind hitting me up with an email please ?  so that I don't lose track?

nielsrune, 2021-08-10:
> @danielbrunt57 @nielsrune would you mind hitting me up with an email please ? so that I don't lose track?

@adonno I beat you to it :-) See my e-mail of August 8th to adonno@adonno.com, subject "LED seems to be faulty or missing"

adonno, 2021-08-11:
i'm working on a solution by end of the week

0n3-70uch, 2021-08-19:
> Just to add to this, my LED isn't working at all.
> After seeing this post, I opened the lid which has the hole (equal to the bottom one in OP's picture)
> 
> But the LED won't turn on, even when switching on in HA :-(

Hello, I have the same problem. The LED worked fine for just one hour...

hermanops, 2021-08-23:
Hi, I also have this problem. I bought the Tag Reader Sandwich Version and the LED isn't working anymore

Strangehadron, 2021-08-24:
I've made a tag reader, same wiring than the schematics and same code and the led wasn't working either. (a W2812B)

I've comment the line : `default_transition_length: 10ms `to desactivate it and now i can turn on/off and change the color of my led via HA, but the led don't flash when scan a tag.

Using the last version of esphome : 2021.8

nielsrune, 2021-09-01:
> i'm working on a solution by end of the week

@adonno, any update on this one?
As per the picture I sent by e-mail, my LED seems physically missing.

adonno, 2021-09-16:
small update , i have been waiting on parts and they finally arrived i'll update you this weekend

0n3-70uch, 2021-09-16:
> small update , i have been waiting on parts and they finally arrived i'll update you this weekend

Great thank you.

adonno, 2021-09-28:
@danielbrunt57 i am unable to get back to you via email it seems i'm blocked ?

adonno, 2022-01-25:
everyone should have gotten their fixed devices?

danielbrunt57, 2022-01-25:
No sign of mine yet...

adonno, 2022-01-30:
yeah just noticed it's still in transit.  it was sent on Dec 8. 
i will have to check with postal services there are several packages stuck

adonno, 2022-04-11:
this time they finally arrived. :-D

closed 2022-04-11 by adonno