adonno/tagreader #125: tagreader device shows red status in the ESPhome dashboard

paulius2k, 2021-07-19:
The device works fine, however it shows with a red status line in the ESPhome dashboard.

I have this option in the yaml:
```
name_add_mac_suffix: true
```
The device's name was registered as `tagreader-xxx` on my router.
However, ESPhome tries to connect to another name, here is the log:
```
WARNING Error resolving IP address of tagreader.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 1 seconds
```
i.e. for some reason it tries to connect to the `tagreader.local`, not the `tagreader-xxx.local`.

Changing the option to:
```
name_add_mac_suffix: false
``` 
solves the problem.

How can I make it work with the option set to `true`?

zuidwijk, 2021-07-19:
Because of the ```name_add_mac_suffix: true``` it add's a part of the mac-address to the name. So, yeah if you created it as tagreader.local, than it can't connect.

If you've got one reader you can remove the ```name_add_mac_suffix: true``` ... if you've got multiple, than just keep one like you have now and update your readers via usb.

zuidwijk, 2021-07-19:
... or make for each reader an unique item...

1. reader1.local
2. reader2.local
3. reader3.local
4. etc.

paulius2k, 2021-07-20:
ok, closing.

closed 2021-07-20 by paulius2k