adonno/tagreader #126: after ESP8266 restart HA settings "buzzer enabled" and "LED enabled" get reset

paulius2k, 2021-07-20:
I have settings "buzzer enabled" and "LED enabled" in my HA set to "true".
But after ESP8266 restart those settings get reset to "false".
Is there any way to set them permanently?

nielsrune, 2021-07-21:
I have done it with an automation to avoid having to re-upload the code to the chip.
This is my automation for the buzzer only. Add LED if you want that too:
```
alias: Tagreader startup script
description: Set tagreader buzzer on when HA is started
trigger:
  - platform: homeassistant
    event: start
  - platform: state
    entity_id: binary_sensor.tagreader_status
    from: 'off'
    to: 'on'
condition: []
action:
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - type: turn_on
    entity_id: switch.tagreader_buzzer_enabled
    domain: switch
mode: restart

```

paulius2k, 2021-07-22:
Thank you, @nielsrune. This does fix the problem partly, just like a periodic script (e.g. every hour) would. However, the settings get reset when the **esp8266** is rebooted, which has no relation to the HA restart event.
Is there a way to install those settings into the esp8266?

nielsrune, 2021-07-22:
I haven't tried yet, but it seems trivial, cfr #88

paulius2k, 2021-07-22:
@nielsrune  - thank you!  
including the lines from #88 into tagreader.yaml really solves the problem :)
```
    - switch.turn_on: buzzer_enabled
    - switch.turn_on: led_enabled
```

closed 2021-07-22 by paulius2k