adonno/tagreader #127: 3D printing questions

sharebear, 2021-07-31:
Hi!

I decided to try and make a couple of these with locally sourced parts and used it as an excuse to finally invest in a 3D printer ;)

My first issue is due to the fact that I have the Lolin V3.1.0 board, the USB and the switch aren't in the same place meaning that, no matter what I try the USB port doen't match up to the hole. Anyone else stumbled across this (I see in the issues others have used the Lolin board)? Any tips for a fix?

Next when researching the issue I found the V4.1 STL files on Thingverse that appear updated compared to the files in this repository. Trying this I have the same problem with the misaligned USB and there doesn't appear to be enough supports for the two boards. Am I right in guessing that this is to be used with the PCB here https://shop.adonno.com/products/tag-reader-pcb ?

What's the best solution in my situation? Should I just design my own case? Are there any source files to start from?

Perhaps add a note in the readme here regarding the relationship between the STLs in this repo and the V4.1 files on Thingverse?

adonno, 2021-08-10:
Hi 

Yes the design wasn't made for a V3 board and thus won't fit into the case for now I haven't had the time to change the design to accept a v3 as of yet I might work on it in the coming weeks though

About the V4.1 case, that's also exact it's made for the PCB version and thus isn't on Github. 
I am not keen on publicly sharing the source files, because it's a mess you might hit me up on discord if you want to try to change something

sharebear, 2021-08-10:
I brute forced the installation for now. I just clipped away a little plastic that was blocking the USB port and I'm not so bothered about the reset button not being accessible (easier to pop open the lid if the need arises, than finding something to stick through the hole).

Thanks for the project, it's been working nicely so far, the main weakness appears to be my soldering skills (second one I constructed appears to be fully working).

adonno, 2021-09-17:
glad to know everything worked out .

closed 2021-09-17 by adonno