adonno/tagreader #130: Is my reader working properly?

Blai89, 2021-09-05:
When i scan the key chain i'm getting this messages in the log:

[D][pn532:277]: Mifare classic
[E][pn532.mifare_classic:093]: Authentication failed - Block 0x04
[D][tagreader:161]: No NDEF
[D][pn532:159]: Found new tag 'XX-XX-XX-XX'
[D][pn532:292]: Waiting to read next tag

Anyway the tag appears on the tags pages in my HA. It's something wrong?

Another thing that I don't know if it is working well is the led and the buzzer, when are they going to turn on? i've suposed every time i put the nfc key chain but this does not happen.

Thank you guys

Blai89, 2021-09-10:
can someone help me? Thanks

adonno, 2021-09-17:
Hello 

Yes, your reader is working properly. 

The LED and Buzzer can be turned on via the HA interface but will have to be turned on again after de device restarts. IF you would like to change that, then you'll have to change the code via ESPHome 
[image]

closed 2021-09-17 by adonno