adonno/tagreader #142: Doesn't work with either of my esp32 devices

colemickens, 2021-12-02:
I have a nodemcu esp-32s and a devkit esp32 and both don't work.

The thing is, I used to have a esp32 that did work with the pn532 and I think it was one of these devices. I'm not sure if I upgraded esphome and it broke, or if something bad happened to this board?

```
[14:47:33][I][i2c.arduino:156]: Performing I2C bus recovery
[14:47:33][C][pn532:018]: Setting up PN532...
[14:47:33][D][pn532:033]: Found chip PN532
[14:47:33][D][pn532:034]: Firmware ver. 1.6
[14:47:33][E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
[14:47:33][E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
[14:47:33][E][component:112]: Component pn532 was marked as failed.
```

sometimes the version IS able to be retrieved, but often not.

colemickens, 2021-12-02:
```
[14:57:05][I][app:029]: Running through setup()...
[14:57:05][I][i2c.arduino:156]: Performing I2C bus recovery
[14:57:05][E][i2c.arduino:172]: Recovery failed: SCL is held LOW on the I2C bus
[14:57:05][E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
[14:57:05][C][pn532:018]: Setting up PN532...
[14:57:05][E][pn532:022]: Error sending version command
[14:57:05][E][component:112]: Component pn532 was marked as failed.
```

some things I'm seeing online say that additional components are needed? Is the circuitry still the same if I don't have the speaker wired in?

adonno, 2021-12-23:
this is weird , can you send me a picture of your setup, please?

adonno, 2022-01-25:
closed due to no feedback

closed 2022-01-25 by adonno