adonno/tagreader #157: Slow boot time

JohnMcLear, 2022-03-16:
Hi, I'm experimenting with these units (I purchased two and thanks btw!) and noticed that the "lead time to scan" IE the time from powering on the device to HA being ready to interact is ~10 seconds.

I'm used to ESP32 devices being online a lot sooner, is there some setting/flag/config/bug we can address to speed up "lead time to scan"?

Thanks!

adonno, 2022-03-16:
Yes that is quite normal, tag reader can only read tags once it' properly connected to HA that might take give it or take 10 secs

It reads the data and immediately sends it to HA. Even it would store it and send it yo HA once it connects both parties need to be properly connected to work.

JohnMcLear, 2022-03-19:
Do we know any of the data/breakdown of why this connection takes so long?  IE from power on, how much of it is initiating Wifi, then subsequently connecting to HA and how much initiating PN532.  Is it possible to parallelize these tasks?

jesserockz, 2022-03-19:
HA connects to the ESPHome devices, not the other way. This is triggered by the mDNS broadcast that is sent out once the device connects to wifi.

Things like manual ip address and wifi fast connect can speed the wifi connection up allowing the boot sequence to be a bit faster.

The initialisation of the PN532 is actually done very quickly and early on in the boot, but a scanned tag cannot be sent to HA until it connects.

JohnMcLear, 2022-04-02:
Using ESPHome I set config as

```
substitutions:
  name: tagreader-017dc7
packages:
  adonno.tag_reader: github://adonno/tagreader/tagreader.yaml
esphome:
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Optional manual IP
  manual_ip:
    static_ip: 10.0.0.243
    gateway: 10.0.0.253
    subnet: 255.0.0.0
```

and I'm still experiencing ~12 second power to ready times.

including ``fast_connect true`` I don't see any improvement.

Am I missing something or doing something wrong because I'm not seeing the expected time reductions as per the comments in this thread.  Thanks so much for the comments so far btw, very useful and informative :)

adonno, 2022-04-03:
I could attempt modifying the code a bit for you to have TagReader store the data of the scanned tag in a "variable" of some kind an have HA read it once it's connected to Tag Reader ( it will still take 12 seconds though for HA to get the value) 

What behavior do you expect and what is the use-case?

JohnMcLear, 2022-04-03:
@adonno thanks for the offer but storing in a variable wouldn't work for me as I need HA communication within my time constraints.

The behaviour that I expect is ready to go within a few seconds and the use-case is to wake the NFC reader with a different piece of hardware/interrupt so to operate within lower power constraints.

adonno, 2022-04-11:
unfortunately it won't boot faster. this isn't a Tag Reader constraint as tags could be read within a second

closed 2022-04-11 by adonno