adonno/tagreader #158: Invalid object error

matat60, 2022-03-20:
I am always having this same issue when I try to "adopt" my new tagreader:

`
INFO Reading configuration /config/esphome/tagreader-49ee0c.yaml...
INFO Updating https://github.com/adonno/tagreader.git@None
Failed config

packages: [source /config/esphome/tagreader-49ee0c.yaml:4]
  
  error: invalid object 100755 69653e652a2f12b45fa608dafe9eec989e61b75f for 'STLs/Cover-circles_no_logo.stl'
  Cannot save the current worktree state.
  adonno.tag_reader: github://adonno/tagreader/tagreader.yaml [source /config/esphome/tagreader-49ee0c.yaml:4]
`

adonno, 2022-03-20:
I have deleted an existing device and tried re adding it and it seems to be working fine , 

can you add a screenshot please ?

matat60, 2022-03-21:
[image]
Here my issue

closed 2022-03-27 by matat60