adonno/tagreader #159: Tag read is resetting reader on latest version

rbray89, 2022-03-21:
Not much to say. Attempting to scan a car results in the reader resetting. This is on "official" hardware from the store.

adonno, 2022-03-21:
Have you tried installing the latest version? 
They have worked on the led issues

rbray89, 2022-03-21:
> Have you tried installing the latest version? They have worked on the led issues

Yeah. Latest version was when I first noticed the issue.

Kept rolling back until the issue was resolved.

So something between the two commits broke my reader.

docmattman, 2022-03-30:
I'm seeing the same issue.  Constant resetting on the device and always never works to scan a tag.

m-paulus, 2022-04-03:
I'm having an issue like this as well. 

Sometimes it creates a new tag in HA with the UID of the tag instead of reading the contents of the tag.

Other times it just reboots when it detects a tag. It will do that several times and then start working again, like it needs to "warm up"

adonno, 2022-04-11:
can you connect Tag reader to you pc and using https://web.esphome.io get the logs when it crashes ? 
do not use logs via WIFI

seedstohr, 2022-05-01:
I am having the same issue. Here are the logs.
[logs.txt](https://github.com/adonno/tagreader/files/8599972/logs.txt)

Lampyridae1984, 2022-05-29:
Yeah, I have the same issue with my tag reader (also bought online pre-assembled). Updated it a couple of hours ago, [here](https://adonno.github.io/tagreader/), for the first time since purchase. I'll upload my log tomorrow. ✌️

adonno, 2022-05-30:
> I am having the same issue. Here are the logs.
> [logs.txt](https://github.com/adonno/tagreader/files/8599972/logs.txt)

Just out of curiosity are you using the provided card / chip ? 
have you installed the latest update since it seems it's running one from march ?

adonno, 2022-05-30:
> Yeah, I have the same issue with my tag reader (also bought online pre-assembled). Updated it a couple of hours ago, [here](https://adonno.github.io/tagreader/), for the first time since purchase. I'll upload my log tomorrow. ✌️

plese do so, this is wierd, and must be software realted at least i hope so

Lampyridae1984, 2022-05-30:
Just for the record, I'm using both the provided cards/chips and my own. I'm also running the latest and greatest software/updates for Home Assistant OS, the ESPHome addon and the tag reader.

When I use a card I now get this error, but it generally still works:
`[E][pn532.mifare_classic:095]: Authentication failed - Block 0x04`

But when I use the chip however, I get this message and the tag reader reboots:
`[E][nfc.ndef_message:078]: Too many records. Max: [4]`

Here is the log:
[logs.txt](https://github.com/adonno/tagreader/files/8796979/logs.txt)

adonno, 2022-05-30:
ok so it crashes only when you read the chip ?  
in that case maybe erase the chip if it isn't already written. 

there is a service : 

[image]

could you try again after cleaning the tag ?

Lampyridae1984, 2022-05-30:
> ok so it crashes only when you read the chip ?
> in that case maybe erase the chip if it isn't already written.

I'm way ahead of you. 👍 As stated above, the very first time I updated the tag reader was yesterday. Before that, I had neither an exposed "clean" nor "write random" function. Anyway, you assumed correctly. The tags were somehow corrupted and cleaning them did the trick.

What is more unsettling is the fact that I've so far also found +10 cards (among my roughly 100) that were corrupted as well. All of them have been used before without problem. Is it normal for these cheap RFID/NFC tags and cards to get corrupted?

adonno, 2022-05-30:
I can't tell if it's normal or not usually they shouldn't be any issue because they get only read and writing by the tagreader or phones only happens on demand afaik

Lampyridae1984, 2022-05-30:
> I can't tell if it's normal or not usually they shouldn't be any issue because they get only read and writing by the tagreader or phones only happens on demand afaik

Fair enough. Thanks for your help, sorry for worrying you unnecessarily. 🙇‍♂️

Quick question though, does the newer square shaped tag reader (with USB-C) have a better WiFi module then the older rectangular one (with micro-USB)? I own one of each. My older one is (has almost always been) really stingy/grumpy when trying to connect to my WiFi, even though it's almost next to my router or my access point.

adonno, 2022-05-30:
both are basically the same, meaning you could update the micro-usb one as well maybe that could solve some issues ?

Lampyridae1984, 2022-05-30:
> both are basically the same, meaning you could update the micro-usb one as well maybe that could solve some issues ?

Already tried that, didn't help. 🤷‍♂️

adonno, 2022-05-30:
Both are based on esp8266 
I have no idea 🤷🏻‍♂️ sorry no clue how i can help further. Removit from the accespoint interly reflash it and retry ?

Lampyridae1984, 2022-05-30:
> Removit from the accespoint interly reflash it and retry ?

Pretty much tried it all, doesn't change a thing. One of those things that can happen with these inexpensive controllers I guess. Thanx anyway. 😊

I'll decide later between replacing the esp8266 or just put some extra € in your pocket. 😉

adonno, 2022-06-06:
glad we fixed the issue

closed 2022-06-06 by adonno

m-paulus, 2023-01-23:
I am still having this issue after cleaning all of my tags. 1 out of every 10 scans works, but most of the time the tagreader reboots every time I scan a tag.

I will plug it into a computer to get the logs.

Lampyridae1984, 2023-01-23:
I had the same issue some time ago. For some reason a couple of my tags where corrupted and that made the reader crash and reboot.

m-paulus, 2023-01-24:
@Lampyridae1984 yes I bought my reader from Adonno's store. I use the "clean tag" function on the reader to erase my tags then I used my android phone to rewrite them. I have a few different kinds, like plastic ones and stickers. The reader reboots almost every time I scan any of my tags. Sometimes it works, then it will work for a few more consecutive scans, then start rebooting again on every scan. 

I haven't had a chance to plug it into a computer yet to get the logs. I will try tonight.

rbray89, 2023-01-24:
The other thing, is that we don't see this issue on the earlier commit.
I've purposefully held back updates as they break things.

m-paulus, 2023-01-25:
Here are logs from the tagreader connected to my computer. 

After a few successful scans, some scans of the same tag fail. 

Sometimes it just reboots immediately. Other times there is an exception (panic/abort/assert) trace.

One thing I have noticed is that this seems to only happen if I have the **buzzer enabled**. When the scan fails, the buzzer sound is drawn out longer. Could it be a hardware problem with the buzzer?

```
[D][pn532:283]: Mifare ultralight
[D][light:035]: 'TagReader LED' Setting:
[D][light:040]:   Color mode: RGB
[D][light:046]:   State: ON
[D][light:050]:   Brightness: 100%
[D][light:057]:   Red: 0%, Green: 100%, Blue: 0%
[D][light:077]:   Flash length: 0.5s
[D][pn532:162]: Found new tag '04-79-9A-0A-00-00-01'
[D][pn532:166]:   NDEF formatted records:
[D][pn532:168]:     U - https://www.home-assistant.io/tag/a9252b07-0c5e-4dd8-9066-58fde57d718a
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android.minimal
[D][pn532:295]: Waiting to read next tag
[D][tagreader:271]: Found Home Assistant tag NDEF
[D][rtttl:038]: Playing song success
[D][rtttl:094]: Playback finished
[D][pn532:283]: Mifare ultralight
[D][light:035]: 'TagReader LED' Setting:
[D][light:040]:   Color mode: RGB
[D][light:046]:   State: ON
[D][light:050]:   Brightness: 100%
[D][light:057]:   Red: 0%, Green: 100%, Blue: 0%
[D][light:077]:   Flash length: 0.5s
[D][pn532:162]: Found new tag '04-14-95-0A-00-00-01'
[D][pn532:166]:   NDEF formatted records:
[D][pn532:168]:     U - https://www.home-assistant.io/tag/2e854f28-7343-424d-adf7-1e7231110675
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android.minimal
[D][pn532:295]: Waiting to read next tag
[D][tagreader:271]: Found Home Assistant tag NDEF
[D][rtttl:038]: Playing song success
[D][rtttl:094]: Playback finished
[D][pn532:283]: Mifare ultralight
[D][light:035]: 'TagReader LED' Setting:
[D][light:040]:   Color mode: RGB
[D][light:046]:   State: ON
[D][light:050]:   Brightness: 100%
[D][light:057]:   Red: 0%, Green: 100%, Blue: 0%
[D][light:077]:   Flash length: 0.5s
[D][pn532:162]: Found new tag '04-FC-8C-0A-00-00-01'
[D][pn532:166]:   NDEF formatted records:
[D][pn532:168]:     U - https://www.home-assistant.io/tag/074fbd47-a312-4afa-a0fe-389512599bc9
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android.minimal
[D][pn532:295]: Waiting to read next tag
[D][tagreader:271]: Found Home Assistant tag NDEF
[D][rtttl:038]: Playing song success
[D][rtttl:094]: Playback finished
[D][button:013]: 'Clean Tag' Pressed.
[D][light:035]: 'TagReader LED' Setting:
[D][light:040]:   Color mode: RGB
[D][light:046]:   State: ON
[D][light:050]:   Brightness: 100%
[D][light:057]:   Red: 100%, Green: 65%, Blue: 0%
[D][light:084]:   Transition length: 1.0s
[D][pn532:299]: Waiting to clean next tag
[D][rtttl:038]: Playing song write
[D][rtttl:094]: Playback finished
[D][pn532:173]:   Tag cleaning...
[D][pn532:177]:   Tag cleaned!
[D][pn532:295]: Waiting to read next tag
[... 791 lines cut ...]
[W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 5 restarts
[C][api:138]: API Server:
[C][api:139]:   Address: tagreader-e29725.local:6053
[C][api:143]:   Using noise encryption: NO
[C][improv_serial:032]: Improv Serial:
[C][wifi_info:009]: WifiInfo IPAddress 'TagReader IP Address'
[C][wifi_info:009]:   Icon: 'mdi:wifi'
[C][wifi_info:011]: WifiInfo SSID 'TagReader Connected SSID'
[C][wifi_info:011]:   Icon: 'mdi:wifi-strength-2'
[D][api:102]: Accepted 192.168.13.178
[D][main:426]: API is connected!
[D][rtttl:038]: Playing song success
[D][light:035]: 'TagReader LED' Setting:
[D][light:040]:   Color mode: RGB
[D][light:046]:   State: ON
[D][light:050]:   Brightness: 100%
[D][light:057]:   Red: 0%, Green: 0%, Blue: 100%
[D][light:077]:   Flash length: 0.5s
[D][switch:013]: 'TagReader Buzzer Enabled' Turning ON.
[D][switch:013]: 'TagReader LED enabled' Turning ON.
[D][binary_sensor:036]: 'TagReader Status': Sending state ON
[D][api.connection:918]: Home Assistant 2023.1.7 (192.168.13.178): Connected successfully
[D][rtttl:094]: Playback finished

```

Lampyridae1984, 2023-01-26:
@m-paulus If you haven't already, I would flash the tag reader with the latest yaml config and ESPHome version. But don't do it OTA, use a cable. (ESPHome recently got a new way to partition, if I understood it correctly, which requires cable connection.)

If the issue still persists it might be the hardware or the tags/cards.

m-paulus, 2023-01-26:
@Lampyridae1984 I did flash the tag reader yesterday with the all the latest using a USB cable. Unfortunately it still resets when I scan a tag with the buzzer enabled. With the buzzer turned off, it seems to work fine.

rbray89, 2023-01-26:
I wonder If the buzzer is using too much power and brown-outs the device.
Or causing EMF that is pulling on the reset pin?

m-paulus, 2023-01-26:
When I send service calls to play sound on the buzzer it plays without rebooting. So it only resets when doing all the tag reading, processing, and buzzing at the same time.

rbray89, 2023-01-26:
Is anyone aware of any power settings changes in either upstream libraries
or the adonno repo?

danblah, 2023-02-02:
I'm getting possible conclusive results. Testing with 4 devices, all running the most recent firmware.

With the buzzer enabled on all, they all restart right after scanning (becoming unavailable in hass as well), more often than not, they do successfully send the tag to hass before restarting. When disabling the buzzer, 4 of 4 will scan w/o rebooting. 

Here's my config:

`substitutions:
  name: tag-kids
packages:
  adonno.tag_reader: github://adonno/tagreader/tagreader.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
logger:
api:
  encryption: 
    key: !secret api_key
ota:
  password: !secret esphome_secret`

Here's what I see in ESPHome logs with buzzer enabled:

`[D][pn532:283]: Mifare ultralight
[D][light:035]: 'TagReader LED' Setting:
[D][light:046]:   State: ON
[D][light:050]:   Brightness: 100%
[D][light:057]:   Red: 0%, Green: 100%, Blue: 0%
[D][light:077]:   Flash length: 0.5s
[D][pn532:162]: Found new tag '04-12-56-01-30-3F-03'
[D][pn532:166]:   NDEF formatted records:
[D][pn532:168]:     U - https://www.home-assistant.io/tag/daf77081-0959-43c9-89e8-072f06f1f520
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android
[D][pn532:168]:     android.com:pkg - io.homeassistant.companion.android.minimal
[D][pn532:295]: Waiting to read next tag
[D][tagreader:271]: Found Home Assistant tag NDEF
[D][rtttl:038]: Playing song success

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffff3f0 end: 3fffffd0 offset: 0000
3ffff3f0:  feefeffe feefeffe feefeffe 3ffe971e  
3ffff400:  000000fe 00000000 00000000 00000000  
3ffff410:  00000000 00000000 00000000 3fffa37b  
3ffff420:  3fffefd0 000000fe 3fff1138 00000001  
3ffff430:  00000000 00000000 0000001f 3fffa10c  
3ffff440:  3fffa12c 4021bd08 00000068 4023d97e  
3ffff450:  3fffa12c 4021bd08 00000020 4023d99d  
3ffff460:  3fffa374 4021ab5c 00000068 4023cfdc  
3ffff470:  3fffa0b8 3ffff490 3ffff4e0 4021bd08  
3ffff480:  3fffa124 3fffa3ac 3ffff4e0 4021abc4  
3ffff490:  00000005 3fffa0a4 3ffff50c 4024e250  
3ffff4a0:  00000018 4021a967 3fffa0c4 4024ea84  
3ffff4b0:  00000000 4021ad79 00000010 3fffa714  
3ffff4c0:  3ffff5dc 3ffff540 3fff4d74 40229d4c  
3ffff4d0:  3ffff5dc 3ffff540 3fff4d74 40229d6b  
3ffff4e0:  3fffa374 3fffa37b 3fffa37b 3fffa384  
3ffff4f0:  00000010 0000001e 3fffa33c 4024e250  
3ffff500:  00000018 00000000 00000000 3fffa354  
3ffff510:  00000014 00000014 00000020 401016f0  
3ffff520:  3ffff5dc 3ffff5b0 3fff4d74 40229d4c  
3ffff530:  3ffff5dc 3ffff5b0 3fff4d74 4022a5e5  
[... 166 lines cut ...]
3fffffa0:  3fffdad0 00000000 3fff0800 3fff0814  
3fffffb0:  3fffdad0 00000000 3fff0800 4023d4a8  
3fffffc0:  feefeffe feefeffe 3ffe8618 401004b5  
<<<stack<<<

last failed alloc call: 4021BD08(104)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x4021bd08

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000917b0
~ld
[I][logger:258]: Log initialized
[C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...`

m-paulus, 2023-02-08:
For the time being, I have been calling rtttl service to play the buzzer from home assistant in an automation following a successful tag read. This gives me better feedback because it confirms that home assistant received the scan event and that the tag was recognized. Even if this issue does get fixed somehow, I may actually keep it this way.