adonno/tagreader #160: Can't install with esphome - error "multiple definition of `info`"

emmost, 2022-03-22:
Hello,

When I tried to install throught esphome, I had the error:

```
/home/[...]/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/[...]/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lib/NONOSDK22x_190703/libmain.a(app_main.o):(.bss+0x4): multiple definition of `info'; .pioenvs/tagreader/src/main.cpp.o:(.bss.info+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/tagreader/firmware.elf] Error 1
========================= [FAILED] Took 44.98 seconds =========================
=============== [ERROR] config/rfid-01.yaml ===============
```

Can you help me ?

adonno, 2022-04-03:
so you have esphome add-on?  and when you try to add tag reader to the add-on you get this ?

adonno, 2022-04-11:
i can't help you if you don't reply ?

adonno, 2022-11-01:
no reply in due time

closed 2022-11-01 by adonno