adonno/tagreader #161: Use TagReader with ESPHome & ioBroker

serb91, 2022-03-23:
Hi,
is it possible to the TagReader with ESPHome in ioBroker?
I don't use Homeassistant but i want to try to integrate your project in my smarthome via iobroker :-)

I try to flash my ESP8266 but i get this:

[image]

can you show me what i have to change?
thx!

adonno, 2022-04-03:
I am working on a MQTT version as well 
does iobroker support espHome?

serb91, 2022-04-03:
> I am working on a MQTT version as well
> does iobroker support espHome?

Yes, iobroker support esphome. 
a mqtt version will be also great

adonno, 2022-04-11:
so all i can do with MQTT right now is get the TAG ID no NDEF messages (i could i guess but i need some time ). is that ok ?

adonno, 2022-04-11:
can you try installing the following 
https://github.com/adonno/tagreader/pull/164/files

adonno, 2022-11-01:
no replies

closed 2022-11-01 by adonno