adonno/tagreader #163: Invalid object error 2

matat60, 2022-03-27:
[image]
Here my issue

_Originally posted by @matat60 in https://github.com/adonno/tagreader/issues/158#issuecomment-1073753649_

adonno, 2022-04-11:
sorry for asking but did you have esphome installe dprior and di you have a tagreader already configured ?

adonno, 2022-04-11:
you have to delete it from esphome. 
maybe also flash it from https://adonno.github.io/tagreader

matat60, 2022-04-16:
Having the same error with flashing by chrome.

adonno, 2022-05-30:
do you have already a tag reader installed in esphome ?  
if so delete it and readapt it please

adonno, 2022-06-05:
do you use custom DNS ? or firewalls ?

adonno, 2022-11-01:
no reply

closed 2022-11-01 by adonno